arras-energy/gridlabd-old #14: need to save random state separately from random seed

dchassin, 2018-10-17:
The `global_randomseed` is used as the storage space for the current state of the random number generator.  This makes it impossible to recover the original seed and reset the state in a subsequent simulation.

Recommend adding a new global called `global_randomstate` and setting the `ur_state` in `gldcore/random.c/random_init` to point to that instead of `glolbal_randomseed`.

dchassin, 2018-10-22:
The attached files should be used to test this.

[issue_14-r2.zip](https://github.com/dchassin/gridlabd/files/2503707/issue_14-r2.zip)

aivanova5, 2018-10-24:
Attached is the model file. The randomseed is not picked up from the previous run and on_sync is not working. @dchassin 
[issue_14-r3.zip](https://github.com/dchassin/gridlabd/files/2511597/issue_14-r3.zip)

dchassin, 2018-11-01:
Ok to merge when PR review is complete.

closed 2019-05-06 by aivanova5