arras-energy/gridlabd-old #21: Add support for ISO 8601 Datetime formats

jcald1, 2018-10-23:
GridLAB-D should be able to handle datetime formats such as:

```
2008-09-15T15:53:00+05:00
2018-10-23T17:15:35:00Z
```

dchassin, 2018-11-15:
@jcald1: The full standard is rather complex.  Do we want to implement all of it, or just a subset of the standard? If only a subset, could we identify which parts?

jcald1, 2018-11-15:
@dchassin , in addition to the canonical format above, it would be good to support the UTC offset format described in the link above right below the description of the canonical format.  These two formats themselves have several variations based on precision (precision to the second vs to the millisecond, etc.) and if it's too much work to fully support each of these variations, we could just pick ones that make the most sense.  e.g., precision to the second for non-delta mode with and without the offset (.01Z and .01-08:00), along with the ones for delta mode.

dchassin, 2018-12-02:
This is address by PR #68.

closed 2019-04-25 by aivanova5