arras-energy/gridlabd-old #39: From/to is not being exported to JSON

jcald1, 2018-11-06:
We're trying to get the to/from properties in the JSON dump.  We can see the to connection from the overhead_line to the node in the GLM file, but in the JSON output, we're not seeing any reference to the node at all and no to/from in the node itself, so we're not seeing a way to get the topology from JSON.

GLM File:
```
object overhead_line {
     phases ABCN;
     name "LN5563942-5";
     from "Q16483_CAP";
     to "R42247";
     length 0.005260km;
     configuration "3PH_H-397_ACSR397_ACSR397_ACSR2/0_ACSR";
}

object node {
     phases ABCN;
     name "R42247";
     nominal_voltage 7199.558;
}
```

JSON output.  No references to "R42247" other than the definition:
```
		"R42247" : {
			"id" : "14716",
			"class" : "node",
			"rank" : "5",
			"clock" : "4268336",
			"schedule_skew" : "0",
			"rng_state" : "115093100",
			"heartbeat" : "0",
			"guid" : "0x7dcee4dc7c3e8e67",
			"flags" : "0x100",
			"bustype" : "PQ",
			"busflags" : "HASSOURCE",
			"reference_bus" : "",
			"maximum_voltage_error" : "+0.719956 V",
			"voltage_A" : "+7495.8-33.0728d V",
			"voltage_B" : "+7524.74-153.128d V",
			"voltage_C" : "+7520.39+87.3804d V",
			"voltage_AB" : "+13011.8-3.03646d V",
			"voltage_BC" : "+12996-122.883d V",
			"voltage_CA" : "+13034+117.1d V",
			"mean_repair_time" : "+0 s",
			"frequency_measure_type" : "NONE",
			"sfm_Tf" : "+0.01 s",
			"pll_Kp" : "+10 pu",
			"pll_Ki" : "+100 pu",
			"measured_angle_A" : "+0 rad",
			"measured_frequency_A" : "+60 Hz",
			"measured_angle_B" : "+0 rad",
			"measured_frequency_B" : "+60 Hz",
			"measured_angle_C" : "+0 rad",
			"measured_frequency_C" : "+60 Hz",
			"measured_frequency" : "+60 Hz",
			"service_status" : "IN_SERVICE",
			"service_status_double" : "-1",
			"previous_uptime" : "-1 min",
			"current_uptime" : "+3 min",
			"Norton_dynamic" : "FALSE",
			"generator_dynamic" : "FALSE",
			"GFA_enable" : "FALSE",
			"GFA_freq_low_trip" : "+59.5 Hz",
			"GFA_freq_high_trip" : "+60.5 Hz",
			"GFA_volt_low_trip" : "+0.8 pu",
			"GFA_volt_high_trip" : "+1.2 pu",
			"GFA_reconnect_time" : "+300 s",
			"GFA_freq_disconnect_time" : "+0.2 s",
			"GFA_volt_disconnect_time" : "+0.2 s",
			"GFA_status" : "TRUE",
			"GFA_trip_method" : "NONE",
			"topological_parent" : ""
		},
```

ryanmmmmm, 2018-11-07:
this resolved this issue correct?

https://github.com/dchassin/gridlabd/commit/3c3a17d1efd6b0c0f8d89ce6dfe2adaf37879b18

jcald1, 2018-11-07:
Yes it's resolved in the dchassin/json_dump branch.

dchassin, 2018-11-07:
There's not PR for this yet.  We should probably revisit the original PR because it looks like it had a bunch of stuff in it I don't think we intended to include.  Any suggestions on how to do that?

ryanmmmmm, 2018-11-07:
hmm great question.
 I am nto sure how many unrelated commits there are but you could:

create a branch of master, then git revert xxxx those commits that were accidental and merge that back into master..

then branch master into a new feature/json branch and cherry pick that one fix in the other json branch into there and issue a pull request form the new branch?

ryanmmmmm, 2018-11-07:
there might be a fancier way to do it and u could do git revert in master to the revision right before the json pull request was merged and sortof "start over" too. the history would be a little cleaner in that case

ryanmmmmm, 2018-11-08:
we spoke today and we will just leave changes in master and just push forward with this.  Will wait for David's PR then we should be able to quickly review and merge into master..

dchassin, 2018-11-09:
I'm really not sure about that.  I need to ask PNNL about that.

closed 2018-11-10 by aivanova5