arras-energy/gridlabd-old #55: Ability to record the units of measure

jcald1, 2018-11-16:
When recording a property, we should be able to extract the units of measure.

e.g., for wind speed:

wind_speed = 10
wind_speed.units_of_measure = 'm/s'

dchassin, 2018-11-19:
Finding the unit of a property can be done several ways.  

1. The class itself published the units of the properties, e.g., using json_dump.
1. The command line option --modhelp displays the units (if any).  
1. A recorder may request the value in a compatible unit.  
1. In the mysql module, there is an option to include the units in the output.  

This is not currently an option in the tape module (but it could be). Are you requesting this option for the tape module?

dchassin, 2018-11-29:
The following syntax should generally work for all recorders, multi_recorders, etc.
~~~
object recorder {
  property "x[unit]";
}
~~~
If you find an instance where that doesn't work as expected, please identify it in this issue, and change it from an enhancement to a bug.  If you cannot find any instances where this doesn't work, you can close the issue as "invalid".

jcald1, 2018-12-06:
@dchassin ,  I tried out option 3.

In a collector:

```
object collector {
        name collectorname;
        group class=pole;
        property max(pole_stress);
        property max(critical_wind_speed[m/s]);
        file "temp2.csv";
}

WARNING  [INIT] : randunit() introducing extra randomness to prevent state stagnation and infinite loops
ERROR    [2003-02-01 00:00:00 UTC] : could not convert units from m/s to m, mismatched constant values
ERROR    [2003-02-01 00:00:00 UTC] : aggregate group property 'critical_wind_speed' cannot use units 'm'
ERROR    [2003-02-01 00:00:00 UTC] : collector 39 'max(pole_stress),max(critical_wind_speed[m/s])' contains an aggregate that is not found in the group 'class=pole'

ERROR    [2003-02-01 00:00:00 UTC] : 2003-02-01 00:00:00 UTC: object collectorname stopped its clock (exec)!
ERROR    [2003-02-01 00:00:00 UTC] : exec halted: synchronization failed
FATAL    [2003-02-01 00:00:00 UTC] : shutdown after simulation stopped prematurely
```

In a recorder:

```

         object recorder {
              property pole_status;
              property pole_stress[pu];
              file "temp.csv";
              interval 3600;
                };

ieee123_pole_vulnerability.glm(16005): starting historical simulation from 2003-02-01 00:00:00 to "2003-02-01 00:15:00"
ERROR    [2003-02-01 00:00:00 UTC] : recorder: property 'pole_stress[pu]' not found
ERROR    [2003-02-01 00:00:00 UTC] : recorder 38 'pole_status,pole_stress[pu]' contains a property of pole 37 that is not found
```

I got a similar error in a multirecorder

aivanova5, 2018-12-19:
@dchassin Confirmed the error. 

Notes from test file - using node_8 as demo case: 
- recording equipment_height without units in tape.recorder - NO error
- recording equipment_height[ft] in tape.recorder - ERROR 
- recording equipment_height[m] in tape.recorder - ERROR 
- recording equipment_height[ft] in multi_recorder - NO error
- reducing property list to one line - error remains
- used different notations (ie with and without quotes) - error remains 

[ieee123_pole_vulnerability.glm.zip](https://github.com/dchassin/gridlabd/files/2696006/ieee123_pole_vulnerability.glm.zip)

dchassin, 2019-02-07:
This is not easily done at this time.

dchassin, 2019-02-07:
Please check that the syntax and usage is correct.  Collectors do not support unit conversion.

closed 2019-02-07 by dchassin