arras-energy/gridlabd-old #73: Support 2 Modes - With Failure/Fault and Without

jcald1, 2018-12-05:
The user should be able to run the simulation in 2 modes: 1) no failure/fault occurring when the pole stress is reached and 2) poles breaking when the pole stress is reached and causing a fault.

The first mode would be just to get an an idea of which poles are the most vulnerable.  This option should be able to be specified either when running in server mode or directly in the GLM file.

For the first mode, the fault would not occur and pole stress could potentially go beyond 1.  Whether the pole status should go to "OUT_OF_SERVICE" needs to be determined.  On the one hand if there is no failure then the status should be OK. On the other we need a way to tell whether a pole should have failed.

dchassin, 2018-12-11:
This should be determined by whether the reliability module is loaded.  In the absence of the reliability module, faults should be noted but not implemented.

dchassin, 2018-12-13:
Adding this to worklist for PR #63.

closed 2020-03-27 by aivanova5