arras-energy/gridlabd-old #92: [Creating] Python code to process JSON

jcald1, 2018-12-17:
Provide the ability to export a JSON file to GLM.

Features:
Handle reading JSON and exporting to GLM

Sanity Check

Handle reading GLM and exporting to JSON

Add default values to JSON

Keep the same options in the same options in the target and source (save everything/legacy, minimal, no globals, no macros, no defaults, no internals)

Handle default values when converting to GLM (If a value in the JSON is the same as the default, the GLM file should not store it) - What happens if the default value changes in a new GridLAB-D version and we import an older JSON file?  The check would be made against the old default value which was stored in the JSON along with the actual value?

Should we support options for saving JSON?

dpinney, 2018-12-19:
My understanding of the data flows this will enable:

[image]

Key new capabilities are the safe interaction with the GridLAB-D data model via scripts and GUIs. It solves 2 issues with interacting with .glm files: (1) writing custom serializers/de-serializers and (2) requires handling syntax in the .glm for arbitrary computation plus a macro system.

dchassin, 2019-02-07:
@aivanova5, please add the graph functions to the library.

closed 2021-04-27 by dchassin