arras-energy/gridlabd-old #99: Power in IEEE123 swing bus is much lower than power in line going to a pole

jcald1, 2018-12-22:
In a recent simulation run of the IEEE123 pole vulnerability model, the lines to the poles have a lot more power than the substation swing bus.

e.g.:

first time step
+20964.4 substation (measured real power)
+1.10552e+06 pole (power out real)

last time step
+14474.3 substation (measured real power)
+761657 pole (power out real)

There is no solar in this model and both measurements are in W according to modhelp.

aivanova5, 2018-12-23:
@dchassin I'm seeing the same behavior. Tracing the feeder, I noticed that substation is only connected to 3 loads, so it seems that poles are on a different circuit.

dchassin, 2019-02-07:
The powerflow calcs appear to be working as expected.

closed 2019-02-07 by dchassin