arras-energy/gridlabd-old #138: Python Module Enhancements

aivanova5, 2019-02-15:
This issue is carried over from Merge with Issues (#100 )

## Current issues
1. The `reset` method is not yet implement due to lack of free calls in gldcore after run is completed.
1. `get_transform()` is not yet implement
1. Unit conversion for complex values not yet implemented.
1. Class event handlers cannot be added.
1. There is no way to implement a new GridLAB-D class in python yet. The module needs to define a base class `GldObject` to do that.
1. Object event handlers are only called if there is an event handler already defined for the object's class.
1. Output streams are not flushed immediately so sometimes the messages from different stream appear on stdout and stderr in a different order than they were received.
1. The function `set_global()` does not work before the simulation is started because the core hasn't initialized yet.

aivanova5, 2019-02-15:
To build on docker run commands for python3 install and ncurses install: 
~~~
%host yum install centos-release-scl
%host yum install rh-python36
%host python —version
%host scl enable rh-python36 bash
%host yum install ncurses-devel
~~~

dchassin, 2021-12-28:
1. Automate python3 install done
2. Automate ncuses install done