arras-energy/gridlabd-old #152: Version info is not correct when branch is ahead of remote

dchassin, 2019-03-08:
Some cases do not seem to be handled correctly by `utilities/build_number`.  The code needs a review and possible rewrite so it handles all the known cases.

1. If HEAD is same or behind remote, then display remote `URL/commit/commit_id`
2. If HEAD is ahead of remote or local changes made, then display `user@host:path:commit_id`

jongoncalves, 2019-03-08:
@dchassin I can probably pick up something like this if you don't mind. Seems pretty straightforward.

dchassin, 2019-03-08:
The script is kind of messed up because it's evolved from the old sourceforge/svn code. It might be good to rewrite it after clarifying exactly all it's supposed to do.

dchassin, 2019-04-25:
I think we need to clean up version so it's more consistent and predictable.  Here's a suggestion.

1. make version always show the same thing, e.g., the name, version and build, but nothing else, i.e.,
~~~
host% gridlabd --version
GridLAB-D 4.2.0-190425
~~~

2. add a new command that gives more information about the build, e.g.,
~~~
host% gridlabd --version_info
application: GridLAB-D
version: 4.2.0
build: 190425
repo: https://github.com/gridlab-d/gridlab-d
branch: master
commit: 5c0e4ce76f25aa698082840540da8e5040552150
platform: Darwin
options: mysql
~~~
We should also include a global variable to affect the format of the output, e.g., `version_format=raw`, as shown above, or `version_format=json` for a JSON entity, or `version_format=xml` for an XML entity.

closed 2019-05-01 by aivanova5