arras-energy/gridlabd-old #169: Slow computing times running on Docker

aivanova5, 2019-03-20:
Running GridLAB-D validation gives inconsistent timing results on docker. 

When docker has been restarted validation time with -T 3: ~ 204s 
After multiple runs (inconsistent as to what event reduces the time) in the same image the time increases with -T 3: ~1000s 
No timing issues found through the host system or VMware Centos7

When running validation and exiting mid-run (killing the process), Segmentation fault occurs when re-running validation. Executing validation command again - allows to validate at ~200s time.

aivanova5, 2019-03-20:
Attempts for slowing down docker without a slow down of validation: 
1. Running make install again between validations 
2. Killing the validation process and following by validation 
3. Exiting the image and starting a new one

aivanova5, 2019-03-20:
Found the problem occurring when the computer goes to sleep and then used again. Work-around is through re-starting the docker for now.

dchassin, 2019-04-04:
Good discussion on this topic: https://hackernoon.com/another-reason-why-your-docker-containers-may-be-slow-d37207dec27f

dchassin, 2019-10-21:
Some evidence has emerged recently that now centos:7 is slow all the time.

dchassin, 2020-04-03:
Suggestion on this: modify `exec_clock()` so it does not call `ftime()` when the profiler is off, and modify all the autotests so they don't use the profiler. Remember to add a caveat to the global profiler docs to note that performance may be affected by use of the profiler.

closed 2021-12-28 by dchassin