arras-energy/gridlabd-old #231: make clean and make distclean are not working

aivanova5, 2019-06-11:
~~~
make clean 
make distclean
~~~

Stopped working... error as follows: 

~~~
[root@cdca902d1410 gridlabd]# make distclean
 rm -f gridlabd.bin
test -z "gldcore/build.h origin.txt" || rm -f gldcore/build.h origin.txt
rm -rf .libs _libs
rm -rf assert/.libs assert/_libs
rm -rf climate/.libs climate/_libs
rm -rf commercial/.libs commercial/_libs
rm -rf generators/.libs generators/_libs
rm -rf gismo/.libs gismo/_libs
rm -rf gldcore/link/engine/.libs gldcore/link/engine/_libs
rm -rf gldcore/link/matlab/.libs gldcore/link/matlab/_libs
rm -rf gldcore/solvers/.libs gldcore/solvers/_libs
rm -rf market/.libs market/_libs
rm -rf mysql/.libs mysql/_libs
rm -rf powerflow/.libs powerflow/_libs
rm -rf reliability/.libs reliability/_libs
rm -rf residential/.libs residential/_libs
rm -rf tape/.libs tape/_libs
rm -rf tape_file/.libs tape_file/_libs
rm -rf tape_plot/.libs tape_plot/_libs
rm -rf third_party/CBLAS/.libs third_party/CBLAS/_libs
rm -rf third_party/jsonCpp/.libs third_party/jsonCpp/_libs
rm -rf third_party/superLU_MT/.libs third_party/superLU_MT/_libs
python3 ./gldcore/link/python/setup.py clean
Traceback (most recent call last):
  File "./gldcore/link/python/setup.py", line 141, in <module>
    version = get_version(),
  File "./gldcore/link/python/setup.py", line 132, in get_version
    with open(path + "/build.h") as f:
FileNotFoundError: [Errno 2] No such file or directory: './gldcore/build.h'
make: *** [python-clean] Error 1
[root@cdca902d1410 gridlabd]# make clean 
 rm -f gridlabd.bin
test -z "gldcore/build.h origin.txt" || rm -f gldcore/build.h origin.txt
rm -rf .libs _libs
rm -rf assert/.libs assert/_libs
rm -rf climate/.libs climate/_libs
rm -rf commercial/.libs commercial/_libs
rm -rf generators/.libs generators/_libs
rm -rf gismo/.libs gismo/_libs
rm -rf gldcore/link/engine/.libs gldcore/link/engine/_libs
rm -rf gldcore/link/matlab/.libs gldcore/link/matlab/_libs
rm -rf gldcore/solvers/.libs gldcore/solvers/_libs
rm -rf market/.libs market/_libs
rm -rf mysql/.libs mysql/_libs
rm -rf powerflow/.libs powerflow/_libs
rm -rf reliability/.libs reliability/_libs
rm -rf residential/.libs residential/_libs
rm -rf tape/.libs tape/_libs
rm -rf tape_file/.libs tape_file/_libs
rm -rf tape_plot/.libs tape_plot/_libs
rm -rf third_party/CBLAS/.libs third_party/CBLAS/_libs
rm -rf third_party/jsonCpp/.libs third_party/jsonCpp/_libs
rm -rf third_party/superLU_MT/.libs third_party/superLU_MT/_libs
python3 ./gldcore/link/python/setup.py clean
Traceback (most recent call last):
  File "./gldcore/link/python/setup.py", line 141, in <module>
    version = get_version(),
  File "./gldcore/link/python/setup.py", line 132, in get_version
    with open(path + "/build.h") as f:
FileNotFoundError: [Errno 2] No such file or directory: './gldcore/build.h'
make: *** [python-clean] Error 1
~~~

aivanova5, 2019-06-11:
If you try to `make install` after running `make clean`, there is an error: 

~~~
g++: error: climate/.libs/climate_climate_la-solar_angles.o: No such file or directory
make[1]: *** [climate/climate.la] Error 1
make[1]: Leaving directory `/usr/local/src/gridlabd'
make: *** [install] Error 2
~~~

dchassin, 2019-07-09:
For now you can force them to work using `make clean -i` and `make distclean -i`.

closed 2019-10-17 by aivanova5