arras-energy/gridlabd-old #232: object-specific precommit function in python version not working

mlamlamla, 2019-06-12:
I am trying to write an object specific precommit function in the python version of GridlabD but it returns an error for different possible return values (t, t_next, True).

To reproduce, checkout my code: https://github.com/mlamlamla/powernet_pyGridlabD_dev and  run "main_test.py".
- I define a market object in IEEE123_BP_2bus_1min.glm with an object-specific precommit function.
-  gridlabd_functions.py defines the object-specific precommit function precommit(obj,t)
- it returns an error for all possible return values

Desired result: The object specific precommit function runs every interval as specified in the object (here for the market object: 300). This should be independent of the global precommit function intervals (here 60 as defined in the players).

@dchassin @aivanova5

aivanova5, 2019-06-17:
Simplified model that fails is here 
[precommit_bug.zip](https://github.com/dchassin/gridlabd/files/3298945/precommit_bug.zip)

aivanova5, 2019-06-19:
Current implementation of precommit does not return a timestamp and returns a binary (where False is continue). This should be fixed for correct operation.

dchassin, 2019-06-19:
I'm going to post a pseudo-fix that allows you to return a timestamp, but doesn't use the timestamp to set the next clock.  We'll figure out how to do that later.  Right now, advise using `gridlabd.NEVER` for continue the simulation, and `gridlabd.INVALID` to stop it.

mlamlamla, 2019-06-24:
This fix works well but we need a consistent fix.

dchassin, 2019-07-09:
Recommend we implement all event handlers to use the same protocol, e.g., called with one timestamp, and return one timestamp.

closed 2019-10-21 by dchassin