arras-energy/gridlabd-old #242: GLM single file converter is not working

aivanova5, 2019-07-01:
Running the command with ieee123 
~~~
   host% gridlabd -D gld_save_options=MINIMAL model/ieee123.glm test.glm
~~~

gives this error: 
~~~
WARNING  [2018-03-01 00:01:00 EST] : gldcore/property.c:property_write(prop='circuit', addr=<house:313>(name='house_1'), string=0x7ffeee166f70, size=1024): no iterator available
ERROR    [2018-03-01 00:01:00 EST] : gldcore/object.c:object_property_to_string(obj=<house:313>('house_1'), name='circuit', buffer=0x7ffeee166f70, sz=1024): unable to extract property into buffer
WARNING  [2018-03-01 00:01:00 EST] : last warning message was repeated 355 times
ERROR    [2018-03-01 00:01:00 EST] : last error message was repeated 355 times
~~~

Running into the same problem with Virtual Islanding, it's possible it's related to having to iterate through houses since output at compile (as attached ) sets the LOADID PHASE COUNT at the beginning and doesn't loop correctly.
[model_single_file.glm.zip](https://github.com/dchassin/gridlabd/files/3347049/model_single_file.glm.zip)

dchassin, 2019-07-01:
The problem is that write method is an error and now all errors cause an exit.  You can either set the ignore_errors global or we can change this so it is a warning instead of an error.

aivanova5, 2019-07-01:
It looks like I was using the wrong variable (I updated the Wiki pages from `gld_save_options` to `glm_save_options` - typo?). The comment above is updated `glm_save_options` gives a set fault. 

Ignore errors wasn't merged into master yet... Could that still be related?

aivanova5, 2019-07-25:
Addressed by #252

closed 2019-07-25 by aivanova5