arras-energy/gridlabd-old #243: Initialization behavior changes slightly from classic GridlabD to pyGridlabD

mlamlamla, 2019-07-06:
In classic GridlabD, global initialization works by running "gridlabd main.glm" where main.glm contains "script on_init python ./gld_init_houses.py;".

In pyGridlabD, it's by calling "python3 main.py" which in turn calls the global on_init(t) function.

This produces slightly different behavior:

In the first case, model initialization itself is done immediately, calculating also characteristics like "cooling_demand" which is then already readable during the initialization process.
In the second case, model initialization is not finalized until on_init(t) is terminated, meaning that characteristics like "cooling_demand" do not get yet calculated and equal +0.0.

This may seem obvious. I wonder if there is an obvious work around, e.g. having object-specific precommits before the global one or if there is another event function which happens between init and start of the model/the first precommit?
I need to read out heating_demand and cooling_demand before the simulation starts. My current work around is to read it out during the first precommit.

@aivanova5 @dchassin

dchassin, 2019-07-09:
Have you tried using object-specific `on_init()` functions?

dchassin, 2019-10-21:
@aivanova5, please look into whether @mlamlamla was able to resolve the issue using `on_init` event handlers at the object level.

closed 2021-12-28 by dchassin