arras-energy/gridlabd-old #245: Islanding is not working

aivanova5, 2019-07-10:
When forcing the model into two isolated network doesn't solve the power flow correctly. 
By opening a switch on a working model the warnings issued are as follows: 

~~~
WARNING  [2018-01-30 00:00:00 EST] : node_001: Newton-Raphson method is unable to converge to a solution at this operation point
WARNING  [2018-01-30 00:00:00 EST] : transformer:xfrm_103BB is at inf% of its rated power value
WARNING  [2018-01-30 00:00:00 EST] : Line:line_node103B_load103B is at 73238430701818167231858625926838964967714699269912777850163712447345139366473337143296.00% of its emergency rating on phase B!
WARNING  [2018-01-30 00:00:00 EST] : Line:line_node103A_load103A is at 265549789246721054467152967904239910810499372107735035566394542137241365594025600491716608.00% of its emergency rating on phase A!
WARNING  [2018-01-30 00:00:00 EST] : transformer:xfrm_103CC is at inf% of its rated power value
WARNING  [2018-01-30 00:00:00 EST] : Line:line_node103C_load103C is at 66933567863753280954500298270254700448083772706585930158677330261951562524069920970299670528.00% of its emergency rating on phase C!
WARNING  [2018-01-30 00:00:00 EST] : transformer:xfrm_103AA is at inf% of its rated power value
WARNING  [2018-01-30 00:00:00 EST] : house_72 - house:372 is outside of ANSI standards (voltage = 2129064378513006724960095806464744301965206990307517715421604051978811253294674625904730786146287616 percent of nominal 120/240)
ERROR    [2018-01-30 00:00:00 EST] : Water heater line voltage for waterheater:365 is too high, exceeds twice nominal voltage.
WARNING  [2018-01-30 00:00:00 EST] : last warning message was repeated 149 times
WARNING  [2018-01-30 00:00:00 EST] : node_001: Newton-Raphson method is unable to converge to a solution at this operation point
WARNING  [2018-01-30 00:00:00 EST] : transformer:xfrm_103BB is at inf% of its rated power value
WARNING  [2018-01-30 00:00:00 EST] : Line:line_node103B_load103B is at 2665496602280164506369694881514332221781481558099070716438390204936642896674789707802443854735474688.00% of its emergency rating on phase B!
WARNING  [2018-01-30 00:00:00 EST] : Line:line_node103A_load103A is at 1252540581745150916970281850767791132323869959887954604074991830349597521832913374723999476133887213568.00% of its emergency rating on phase A!
WARNING  [2018-01-30 00:00:00 EST] : transformer:xfrm_103CC is at inf% of its rated power value
WARNING  [2018-01-30 00:00:00 EST] : Line:line_node103C_load103C is at 1098939021671196384120055628774571780548177712010433940727591172837274269886494428825491058210444012421120.00% of its emergency rating on phase C!
WARNING  [2018-01-30 00:00:00 EST] : transformer:xfrm_103AA is at inf% of its rated power value
~~~

dchassin, 2019-07-24:
Check that the correct modules and flags are set to enable island solutions.  This should be discussed at some length in the SourceForge forum for GridLAB-D.

dchassin, 2019-10-21:
@aivanova5, is this a bug or a modeling error?

dchassin, 2020-04-26:
@aivanova5, can we close this now?

closed 2021-01-25 by aivanova5