arras-energy/gridlabd-old #282: ISO timestamp can't be read as a macro

aivanova5, 2019-08-12:
Can not pass the ISO timestamp as a macro...See examples attached. `test_iso_datetime.glm` is functional however `test_iso_datime_macro.glm` gives a parsing error as: 

~~~
PC95555:autotest alyona_slac$ gridlabd test_iso_datetime.glm
test_iso_datetime.glm(8): expected time value
test_iso_datetime.glm(8): expected clock block closing }
test_iso_datetime.glm(6): load failed at or near 'clock {...'
ERROR    [INIT] : unable to load 'test_iso_datetime.glm': No such file or directory
FATAL    [INIT] : shutdown after command line rejected
~~~
[Archive.zip](https://github.com/dchassin/gridlabd/files/3493696/Archive.zip)

aivanova5, 2019-08-20:
When defining a timestamp as ISO8601 the simulation doesn't start at the defined time... The attached file has start and stop time in 2018 however the simulation starts in 2010....

[Archive.zip](https://github.com/dchassin/gridlabd/files/3522079/Archive.zip)

dchassin, 2019-08-20:
Recommend the following change:
~~~
@@ -1188,8 +1188,8 @@ TIMESTAMP convert_to_timestamp(const char *value)
		S = (unsigned short)s;
		unsigned int ns = (s-S)*1e9;
		DATETIME dt = {Y,m,d,H,M,S,ns,0};
-		TIMESTAMP t = mkdatetime(&dt);
-		return t - (tzh*60+tzm)*60;
+		dt.tzoffset = (tzh*60+tzm)*60;
+		return mkdatetime(&dt);
	}

	/* scan ISO format date/time */
~~~

closed 2019-10-14 by aivanova5