arras-energy/gridlabd-old #299: python environment shutting down after one GridlabD run

mlamlamla, 2019-08-29:
When running GridlabD through the python environment, it seems that the python environment is shut down and the program ended, once a GridlabD simulation is done. So if multiple Gridlabd models should be run sequentially or data be evaluated using Python only after the simulation, that's not possible.

E.g.:

import sys
import gridlabd
gridlabd.command('main.glm')
gridlabd.command('-D')
gridlabd.command('suppress_repeat_messages=FALSE')
gridlabd.command('--warn')
gridlabd.start('wait')
print('THIS CODE WONT BE RUN ANYMORE')

@dchassin @aivanova5

aivanova5, 2019-08-30:
I think this has to do with the lack of re-entrancy. They way around that is introducing a sh script which will run multiple things after gridlabd is done in python. Here is an example: 
Terminal commmand: sh main.sh
main.sh
~~~~
python3 main.py
python other_code.py
gridlabd --version
~~~~

where main.py would contain the script as you have above until the start command: 
~~~
import sys
import gridlabd
gridlabd.command('main.glm')
gridlabd.command('-D')
gridlabd.command('suppress_repeat_messages=FALSE')
gridlabd.command('--warn')
gridlabd.start('wait')
~~~
and then print('code that didn't run') can go into `other_code.py`

dchassin, 2019-09-01:
The inability to run python code after the ‘start’ command is a problem that first appeared when the exit on error condition was fixed for real-time simulations.  A new PR #301 has been submitted to address this issue.

Note that #301 will not address the issue of running multiple models in the same process. That is indeed a reentrancy problem that is still being addressed by the ongoing core restructuring as noted by @aivanova5.

closed 2019-09-03 by aivanova5