arras-energy/gridlabd-old #305: CSV files are not being generated when running a simulation from a GLM converted from a JSON file

jcald1, 2019-09-03:
GLD Version:        gridlabd/slac-master:latest (on 8/30)
GLM version:        virtual_islanding_model.glm in VI_timeseries branch of grip-server on 8/30

Files to Download: https://drive.google.com/file/d/1V8Qu6A5nBi12Yr1EHUCk1sQ0m5acUFMy/view?usp=sharing
https://drive.google.com/file/d/1hxFKaeIEgyd1Nr3v3MmbqRt1kehXgbbA/view?usp=sharing 

Steps to Reproduce:    
     STEP 1  Create the JSON Dump
1.  Extract 1-1__5d0ae64a-6286-4e8f-855d-1bca3c583587.zip to /tmp/grip/simulation_runs/1-1__5d0ae64a-6286-4e8f-855d-1bca3c583587
2.  virtual_islanding_model.glm copied to new_virtual_islanding_model.glm 
3.  From the directory above, execute this command:
docker run --rm --name gridlabd_1_1_new_virtual_islanding_model.glm_5d0ae64a-6286-4e8f-855d-1bca3c583587 -v /tmp/grip/simulation_runs/1-1__5d0ae64a-6286-4e8f-855d-1bca3c583587:/model -v /tmp/grip/simulation_runs/1-1__5d0ae64a-6286-4e8f-855d-1bca3c583587/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model gridlabd/slac-master:latest python3 -u run_gridlabd_main.py -C -W /model -i new_virtual_islanding_model.glm -o new_virtual_islanding_model.json

   STEP 2 Convert JSON to GLM
1.  Extract 2-2__12767862-1409-4df3-836b-189e14008b6f.zip to /tmp/grip/simulation_runs/2-2__12767862-1409-4df3-836b-189e14008b6f
2.  Copy the output from STEP 1 new_virtual_islanding_model.json into this directory and rename to  virtual_islanding_model_input.json 
No changes were made to the file.
3.  From this directory in step 2, execute the following to convert the JSON to GLM.
docker run --rm --name gridlabd_1_2_virtual_islanding_model_input.json_12767862-1409-4df3-836b-189e14008b6f -v /tmp/grip/simulation_runs/2-2__12767862-1409-4df3-836b-189e14008b6f:/model -v /tmp/grip/simulation_runs/2-2__12767862-1409-4df3-836b-189e14008b6f/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model gridlabd/slac-master:latest python3 -u /usr/local/share/gridlabd/json2glm.py -i virtual_islanding_model_input.json -o virtual_islanding_model_output.glm

   STEP 3 Execute the Simulation
1.  From the directory in STEP 2, execute the simulation (using the output from the last step):
docker run --rm --name gridlabd_1_2_new_virtual_islanding_model_output.glm_12767862-1409-4df3-836b-189e14008b6f -v /tmp/grip/simulation_runs/2-2__12767862-1409-4df3-836b-189e14008b6f:/model -v /tmp/grip/simulation_runs/2-2__12767862-1409-4df3-836b-189e14008b6f/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model gridlabd/slac-master:latest python3 -u run_gridlabd_main.py -W /model -i new_virtual_islanding_model_output.glm -o new_virtual_islanding_model_output.json
2.  No CSVs were generated

dchassin, 2019-10-21:
@aivanova5, I assume this is resolved/not a problem.

aivanova5, 2019-10-21:
yes, closing

closed 2019-10-21 by aivanova5