arras-energy/gridlabd-old #307: Class and Include Failures in IEEE123 Model

evgaston, 2019-09-04:
Multiple modules do not appear to be working when running the IEEE123 model, including the recorder class failing, and the solar include failing.  Warning output below: 
[image]

Not sure, but this may be related to a past issue, as described below: 
https://github.com/gridlab-d/gridlab-d/issues/627 

"mhauer (user not in GitHub. This is the sourceforge username): At one point, there was a problem with the included file having stacks of PV arrays wrapped in #if x>n\ macros. There was a problem with individual macros not counting as 'GLM code', and not getting flushed after a 'negative #if' was processed, surpressing the contents. The buffer would then overflow and stop the simulation.

dchassin, 2021-12-28:
@aivanova5, is there anything left here to do? If, not we can close this issue.

dchassin, 2023-09-27:
This does not appear to be a problem anymore. Closing.

closed 2023-09-27 by dchassin