arras-energy/gridlabd-old #308: Assert with Object Group/Class Functionality

evgaston, 2019-09-04:
The current assert statement requires an explicit named object to be it's parent. It would be ideal for an assert to work on a named group of objects or the entire class. 

Example:  the below would evaluate all defined node objects nominal voltage to be within 2+/- of 7199. 

object node {
  object double_assert {
     value 7199;
     within 2;
     target "nominal_voltage";
  };
}

dchassin, 2019-09-07:
This would probably have to be accomplished by not setting the parent object, and instead setting a `group` property to search object for qualifying targets, e.g.,
~~~
object assert {
    group "class=node";  
    value 7199;
    within 2;
    target "nominal_voltage";
}
~~~

dchassin, 2019-09-07:
While we're at it, we should allow the `assert` object to consider how long the value has not been within the specified range, so that momentary deviations can be ignored if desired, e.g.,
~~~
object assert {
    group "class=node";  
    value 7199;
    within 2;
    target "nominal_voltage";
    duration "30 s";
}
~~~

closed 2019-09-25 by aivanova5