arras-energy/gridlabd-old #325: JSON dump fails to dump custom variables correctly

aivanova5, 2019-09-13:
Defining custom powerflow object properties results in corrupt JSON dump for that property. Simplified model attached.
[jsongarbage.zip](https://github.com/dchassin/gridlabd/files/3611987/jsongarbage.zip)

aivanova5, 2019-09-30:
A temporary fix has been issued however as in PR#326 additional fixes are required for robust implementation.

dchassin, 2019-10-21:
Current issues still pending resolution:
1. When one powerflow class is derived from another, gridlabd must know how big the parent class is to calculate the position of property values in in the new class. The way power flow defines classes, the class are all derived from power flow object, node, link, etc. when the module is loaded. So the only way to add super node name to classes is to add them to classes that don’t have child classes (I’ve added an exception to block that if you try). Normally this would be ok to do, even if it’s odd given how parent classes should work. Unfortunately, because node is used both as an instantiated class (meaning it has objects associated directly with), and as an abstract class (meaning it is a parent of instantiated classes), it is not possible to add an extended variable to node without breaking that rule. This underlying problem needs to be fixed so that it's possible to extend abstract classes after they've been used as a parent to another class.
1. If we going to keep this solution permanently (which we really shouldn't), then we should name the new property something more generic because supernode_name is rather particular to GRIP models, and I’d prefer we use something more thought out based on a concept of super nodes that would be more general purpose to future users.

closed 2021-12-28 by dchassin