arras-energy/gridlabd-old #354: Another method output edge case

aivanova5, 2019-09-27:
When using 
~~~
object test:..10 {
	value "type: normal(0,1); refresh: 1 min"; 
}
~~~
the json dump is literal: 
~~~
"objects" : {
		"test_0" : {
			"id" : "0",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "21770",
			"heartbeat" : "0",
			"guid" : "0x292399a2ee7c841f6ddb1dd673a89402",
			"flags" : "0x100",
			"value": "0.420061"
		},
		"test_1" : {
			"id" : "1",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "2647",
			"heartbeat" : "0",
			"guid" : "0x67fb9004153a079e2552edd0daccde2c",
			"flags" : "0x100",
			"value": "1.227954"
		},
		"test_2" : {
			"id" : "2",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "17841",
			"heartbeat" : "0",
			"guid" : "0x2b764549fa9f285b55c9d972e635be79",
			"flags" : "0x100",
			"value": "-1.044053"
		},
		"test_3" : {
			"id" : "3",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "24475",
			"heartbeat" : "0",
			"guid" : "0x140792773bc739b52193ec897a2d1e15",
			"flags" : "0x100",
			"value": "0.221411"
		},
		"test_4" : {
			"id" : "4",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "2683",
			"heartbeat" : "0",
			"guid" : "0x3cd3606125963c252f274bbe47bc655c",
			"flags" : "0x100",
			"value": "-0.045231"
		},
		"test_5" : {
			"id" : "5",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "8092",
			"heartbeat" : "0",
			"guid" : "0x5453ccd3c10833cba9e42361825f882",
			"flags" : "0x100",
			"value": "0.328062"
		},
		"test_6" : {
			"id" : "6",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "1676",
			"heartbeat" : "0",
			"guid" : "0x27cdf921a004864d46748849be30d8ef",
			"flags" : "0x100",
			"value": "-1.959096"
		},
		"test_7" : {
			"id" : "7",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "12734",
			"heartbeat" : "0",
			"guid" : "0x4fe348bfdb762b781abb13f2427141c3",
			"flags" : "0x100",
			"value": "-0.037961"
		},
		"test_8" : {
			"id" : "8",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "25683",
			"heartbeat" : "0",
			"guid" : "0x727fd2330bdcf1c3600b2f8197413ca4",
			"flags" : "0x100",
			"value": "-0.609855"
		},
		"test_9" : {
			"id" : "9",
			"class" : "test",
			"rank" : "0",
			"clock" : "2018-01-01 00:00:00 PST",
			"schedule_skew" : "0",
			"rng_state" : "26085",
			"heartbeat" : "0",
			"guid" : "0x645060fe9d222ea531631edd19ca07fc",
			"flags" : "0x100",
			"value": "0.100833"
		},
~~~

dchassin, 2019-09-27:
Yes, it is indeed literal only.  It is a desired option to save the original information, rather than the current value.  But it isn’t always possible right now.  (Try it with PR #349 with the `ORIGINAL` option enabled for `glm_save_options`.)  We’ll need to support that for all output formats.

dchassin, 2021-12-28:
@aivanova5, is this still an open issue?