arras-energy/gridlabd-old #357: Restructure installation files into a single easily linked folder

dchassin, 2019-09-30:
GridLAB-D install files into several target folders scattered around the filesystem.  This makes it difficult to manage multiple installation.  The installation files need to be repackaged into a single folder with minimal links to the appropriate system files and folders to enable the user to manage multiples versions and change versions easily and safely.

Proposed structure for installation for 4.2 and beyond:
~~~
/usr/local/src/gridlabd-4.2.0-190930 (optional source code)
/usr/local/gridlabd -- install of the active version (link to specific version)
/usr/local/gridlabd-4.2.0-190930 -- install of specific version
/usr/local/bin/gridlabd* -- main gridlabd command (link to /usr/local/gridlabd/bin/gridlabd*)
/usr/local/lib/gridlabd - gridlabd libraries (link to /usr/local/gridlabd/lib)
/usr/local/share/gridlabd - gridlabd data (link to /usr/local/gridlabd/share)
/usr/local/doc/gridlabd - gridlabd docs (link to /usr/local/gridlabd/doc)
/usr/local/include/gridlabd - gridlabd header files (link to /usr/local/gridlabd/include)
~~~

This approach will require a new `--version` option to encode the specific version name formatted as `<application>-<version>_<build>-<branch>-<system>-<hardware>`, e.g.,
~~~
host% gridlabd -V install
gridlabd-4.2.0_190930-master-macos10.14-x86_64
~~~

This approach will require the main `gridlabd` command script to be aware and allow user management of the active version in the current shell.  Note that changing version should be controlled exclusively from the environment, and not by changing the links, which are used only to determine the system default version.

aivanova5, 2019-09-30:
Concurrently, managing python environments may be a good idea. Dealing with things like anaconda path vs the "correct" install path.

dchassin, 2019-10-21:
This is current being addressed by PR #374

dchassin, 2019-12-08:
This is now available in `develop`.

closed 2019-12-08 by dchassin