arras-energy/gridlabd-old #379: Master fails to pass test_pole_NR.glm through validate.py

aivanova5, 2019-10-10:
Problem with validation through `validate.py`. The error is as follows: 
~~~
Running test_pole_NR.glm...
sh: line 1:  3139 Segmentation fault: 11  /usr/local/bin/python3 /Users/alyona_slac/gridlabd/macosx/gldcore/link/python/python_gridlabd.py -W /Users/alyona_slac/gridlabd/macosx/powerflow/autotest/test_pole_NR test_pole_NR.glm > gridlabd.out 2>&1
FAIL /Users/alyona_slac/gridlabd/macosx/powerflow/autotest/test_pole_NR/test_pole_NR.glm exit 139
~~~

aivanova5, 2020-05-05:
We're not validating through python any more, I vote we close this @dchassin

dchassin, 2020-05-05:
Good point. But maybe we should be validating the python libraries.  Right now, there are no tests to ensure that they're coded and built correctly.

dchassin, 2020-05-05:
I just ran the python validation and the result is unsatisfactory:
~~~
Tested.... 707
Failed.... 56
Passed.... 651 (92.1%)
Time...... 338.4 sec
~~~

aivanova5, 2020-05-13:
Ouch, can we use validate.py to track down the `test_pole_NR.glm` intermittent problem since it seems to fail every time through python validation routine?

closed 2021-12-28 by dchassin