arras-energy/gridlabd-old #400: measured_real_energy_delta has questionable output

aivanova5, 2019-10-23:
Couple problems: 

(1) In `group_recorder` the measured_real_energy_delta outputs 0 even if measured_real_energy property is accumulating energy 
(2) In `recorder` the delta energy from the timestamp is not the same as if you were to calculate back the increase in the energy from measured_real_energy property 

Test files and output attached:
[output](https://github.com/dchassin/gridlabd/files/3763783/Archive.zip)

[test_files](https://github.com/dchassin/gridlabd/files/3763781/Archive.zip)

dchassin, 2021-12-28:
@aivanova5, is this related to the use of complex format based on `%g` instead of `%f`?

BTW, the attachments are no longer available.

dchassin, 2023-09-27:
Assuming this is related to output format issues (which is resolved). Closing.

closed 2023-09-27 by dchassin