arras-energy/gridlabd-old #419: Unable to update objects for reliability effect - 24 Hour GRIP Anticipation Simulations Fail

jcald1, 2019-11-11:
Running 24 hour simulations for the GRIP Anticipation scenario results in an error.  I used the `slacgrip/master:191022` Docker image, which uses `gridlabd/slac-master:1910222` as the base image, which uses the `4.2.0-191022` version of SLAC GridLAB-D

Steps to reproduce:
1.  Clone `https://github.com/PresencePG/grip-server`
1.  Copy the grip-server/components/grip_sim_runner/gridlabd/anticipation directory contents from the cloned grip-server folder to a directory where you will run the simulation.
1.  Extract the Anticipation IEEE123 GLM model and weather file from the attachment to the directory above.
1.  In the directory above run: ` docker run --rm --name gridlabd_85_129_new_ieee123_pole_vulnerability.glm_f8230176-1774-4a79-a90c-a32a358035da -v $(pwd):/model -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py -W /model -i new_ieee123_pole_vulnerability.glm -o new_ieee123_pole_vulnerability.json`

The weather file and GLM model are attached.

Command and output from our run that produced the error (the command is slightly different due to directory differences and and additional Docker volumes):

```
+ docker run --rm --name gridlabd_85_129_new_ieee123_pole_vulnerability.glm_f8230176-1774-4a79-a90c-a32a358035da -v /tmp/grip/simulation_runs/129-129__f8230176-1774-4a79-a90c-a32a358035da:/model -v /tmp/grip/simulation_runs/129-129__f8230176-1774-4a79-a90c-a32a358035da/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py -W /model -i new_ieee123_pole_vulnerability.glm -o new_ieee123_pole_vulnerability.json
Setting path to folder:  /model
GridLAB-D 4.2.0-191022
Error(s) found in the model, printing gridlabd.err file contents...
ERROR    [2003-02-01 11:55:00 UTC] : sync_pole(obj=37;node_13): [powerflow/link.cpp:7651] unable to update objects for reliability effects (function 'reliability_alterations' not defined)
ERROR    [2003-02-01 11:55:00 UTC] : 2003-02-01 11:55:00 UTC: object node_13 stopped its clock (exec)!
ERROR    [2003-02-01 11:55:00 UTC] : exec halted: synchronization failed
FATAL    [2003-02-01 11:55:00 UTC] : shutdown after simulation stopped prematurely
FATAL    [2003-02-01 11:55:00 UTC] : environment startup failed: No such file or directory
```
[24-anticipation-error.zip](https://github.com/dchassin/gridlabd/files/3833412/24-anticipation-error.zip)

aivanova5, 2020-04-01:
@dchassin 

Reduced model attached. 
Simply run: `gridlabd anticipation_ieee123_pole_vulnerability_with_str_repl.glm`

Observations: 

- Happens exactly at `2003-02-01 11:55:00 UTC` when wind speed surpasses `39.7667` (see `weather_recording.csv`) 
- suspect node: `node_13` pole object
- When `node_13` pole changed to node, other pole objects do not exhibit this issue 
- gridlabd source code autotest: `test_pole_NR.glm` exhibits Seg Fault at least every 5th time it is executed. 

[anticipation_reliability_issue.zip](https://github.com/slacgismo/gridlabd/files/4417525/anticipation_reliability_issue.zip)

dchassin, 2020-04-04:
The error `sync_pole(obj=37;node_13): [powerflow/link.cpp:7651] unable to update objects for reliability effects (function 'reliability_alterations' not defined)` means exactly what it says: the fault_check object is not defined in the model. 

Try adding the following 

~~~
object fault_check
{
    check_mode ONCHANGE;
    output_filename "fault_check.txt";
    reliability_mode TRUE;
}
~~~

Once you add that, you may another error regarding a line's invalid phase configuration: `NR: A line's phase was flagged as not full three-phase, but wasn't`.  I have no suggestion on that, so hopefully it's easily fixed.

dchassin, 2020-04-04:
One question: the `fault_check` output is a TXT file.  Do you think we should output a JSON file instead?

dchassin, 2020-04-04:
@aivanova5, one question: the `fault_check` output is a TXT file.  Do you think we should support JSON files?

aivanova5, 2020-04-04:
@dchassin JSON would probably be convenient and standardize this.

dchassin, 2020-04-06:
The problem occurs in the current macOS `develop` branch, but it does not occur in the current Docker image.

closed 2020-04-08 by aivanova5