arras-energy/gridlabd-old #420: GRIP Recovery IEEE37 Simulation Doesn't Complete

jcald1, 2019-11-12:
I ran the IEEE37 simulation, which is being used for the Recovery phase of GRIP.  The simulation never completes, even though the JSON dump was created.  The compile flag was set.

GridLAB-D Docker Image: 
```
REPOSITORY             TAG                 IMAGE ID            CREATED             SIZE
gridlabd/slac-master   latest              24eaf77a3c56        3 days ago          1.37
```

Command:
`docker run --rm --name ieee37 -v $(pwd):/model -w /model gridlabd/slac-master:latest  gridlabd -C --debug --version -W /model recovery_ieee37.glm -o recovery_ieee37.json`

Files in attachment:
```
recovery_ieee37.glm (Model)
recovery_ieee37.json (JSON dump)
gridlabd_output.txt (GridLAB-D standard out)
```
[recovery_ieee37.zip](https://github.com/dchassin/gridlabd/files/3837701/recovery_ieee37.zip)

Note that the output didn't include the GridLAB-D version even though I used the `--version` command line option.

Joscha90, 2019-11-12:
The above mentioned ieee37 feeder originates from the tools folder in the original gridlab-d repository:
https://github.com/gridlab-d/tools/blob/master/IEEE%20Test%20Models/37node/IEEE_37node.glm

I just saw now that there is already an issue request in the repository folder regarding this, but it seems like nobody fixed it:
https://github.com/gridlab-d/tools/issues/1

jcald1, 2019-12-18:
@Joscha90, PNNL should keep their models up to date, but on the other hand, a bad model shouldn't cause GridLAB-D to hang, so I'm leaving this issue open for now.

dchassin, 2019-12-18:
Bear in mind that there are some important differences between 4.1 (PNNL version) and 4.2 (SLAC version).  There is no guarantee that a model from 4.1 will work correctly under 4.2 especially if one of them is development. If you want to run the IEEE 37 bus model you must run the model from the same branch.

jcald1, 2019-12-18:
@dchassin , I should mention that I opened this ticket before we found the newer IEEE37 from PNNL's autotest directory.  That new one does work with SLAC GridLAB-D and is the one we'll be working with going forward.  So this issue here isn't really blocking us in GRIP, as it's based on a model we won't be using in GRIP.  It's up to you if you want to keep it open as something worth investigating.

dchassin, 2019-12-18:
SLAC is planning to tie off the project on 12/31/2019.  This is because PNNL cannot easily accept PRs from this fork anymore--the difference is simply too great to handle without significant effort on their part. PNNL recently started their own version 4.2, and therefore cannot recognize the SLAC version 4.2 that has been under development for CEC since last year.  We are currently reviewing the commercialization plan with the CEC, and in the interim the plan is to transfer the project to `slacgismo` and tie off the fork.

aivanova5, 2020-03-27:
@jcald1 @Joscha90 Is this still an issue?

Joscha90, 2020-03-27:
> @jcald1 @Joscha90 Is this still an issue?

Not from our side, because we don't use the model anymore like Jorge mentioned in an earlier comment.

closed 2020-05-13 by aivanova5