arras-energy/gridlabd-old #467: Problems parsing .mag .ang

aivanova5, 2019-12-09:
## Problem description
While trying to extract the magnitude and angle of the complex number the multi-recorder and recorder are both unable to parse a property with `.mag` and `ang`. Note that the real and imag extensions work. 

## Steps to reproduce
See the simplified file attached 

## Expected behavior
Expecting to get the magnitude and angle of the complex number, however an error occurs while running the model.

## System information
* GridLAB-D version: GridLAB-D 4.2.0-191111 (develop) Darwin (this also relevant for docker and master branch)
* Operating system: Darwin vpn-clients-gw2-113-80.slac.stanford.edu 18.7.0 Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64 x86_64

## Supporting information
N/A

## Attachments
[simple_test.glm.zip](https://github.com/dchassin/gridlabd/files/3941929/simple_test.glm.zip)

aivanova5, 2019-12-09:
@jcald1 This is relevant to the question we were discussing today.

dchassin, 2019-12-12:
The way multi-recorder works it can only link the output to an underlying double directly.  It is not yet possible to link to a method of a class. This is why only `.real` and `.imag` work and nothing else works.  To fix this would require reimplementing multi_recorder so that it uses methods instead of pointers to doubles.

dchassin, 2020-03-04:
The preferred way to record parts of complex values now is to use output formatting, e.g., `property measured_voltage_A[:4fM];` instead of `property measured_voltage_A.mag`.

closed 2020-03-04 by dchassin