arras-energy/gridlabd-old #489: GRIP Absorption JSON Dump Causes Unserved Load to Be Incorrect

jcald1, 2019-12-18:
From @sehowerter:

"I’ve generated plots for both of the No Faults or “baseline” runs that were done in the GRIP UI. The values plotted are the difference in both the water heater load and hvac load between the baseline/no faults run I did locally on my computer and the runs in the GRIP UI. (local - UI)
This shows that there is definitely an issue with the water heater demand after the JSON 
conversion. I think the hvac demand differences are more reasonable since they are both positive and negative, probably cancelling out for the most part to not attribute to higher measures of “unserved energy” between the two simulations.
@David Chassin & @Alyona This seems related to the issue I was having with water heater demand getting shut off once the island management / absorption algorithm started managing them, but that wasn’t connected to the JSON conversion process since I was seeing that in my local tests where I do not do the JSON conversion.
Let me know what you think.
...
The Ice case looks less wrong because a higher proportion of the total flexible load is from the HVAC devices (heating houses in the winter). While the Fire case has almost no HVAC load, leading to most of the flexible load being from the water heaters, so when they essentially shut off, the “unserved energy” skyrockets."

From jcald1:
"Jorge Calderon 4:13 PM
@Alyona, @David Chassin, we looked at the results more carefully and in the fire model baseline run (no faults), there is a large amount of unserved energy: https://integration.grip.presencepg.com/absorption/simulations/37.  Could this be caused by issues with the JSON dump?    We ran the simulation with compile flag on and created the JSON dump, converted it to GLM, then ran the new GLM.  I verified that the new GLM didn't have any faults in it.  Do you think it's worth modifying the JSON dump file to take out all properties that were not configured in the original GLM, to isolate whether incorrect default values were exported to the JSON dump?  If that still doesn't work, then could the issue be something more structural like "if" statements in the original GLM not being exported to the JSON dump and could that cause the number of random generator calls to be different from the original GLM run?  Should there only be a single global randomstate variable or should each class (or object) have its own in the JSON dump? (edited) 

1 reply
27 days agoView thread
Jorge Calderon 4:23 PM
@Alyona, @David Chassin, by the way, I added the global random state to a multi_recorder and the value was the same at all time steps and matched the value in the JSON dump.  I figured that would change during the simulation run.
David Chassin 4:25 PM
It's unusual for it to not change, but possible.  It only changes when something uses the global RNG source. There are a very large number of independent random stream, you're only recording the global one. (edited) 
Jorge Calderon 4:34 PM
@David Chassin, I didn't see any other random states in the JSON dump, only the global one.  Is that in itself a problem and could it explain the issue above? (message at 4:13 PM)?
David Chassin 4:36 PM
That shouldn't be an issue.  The internal random states are hidden, but they're all derived from the global state.  But I'm beginning to think that this is in fact the problem itself.  Resetting the global state doesn't reset the other streams.  They should be saved with the object headers, but they might not be.  I'll check with @Alyona tomorrow to see if they're coming back through JSON to GLM and backtrack it from there.
"

dchassin, 2019-12-18:
@aivanova5, is this the problem described is #470?

jcald1, 2020-06-17:
@dchassin , @aivanova5 , closing as the fix has been verified by @sehowerter.

closed 2020-06-17 by jcald1