arras-energy/gridlabd-old #493: GRIP Absorption Baseline Run Fails To Generate Baseline When Model is Run From Converted JSON Model

jcald1, 2019-12-21:
When converting an Absorption model to JSON, then back to GLM (to be able to update the weather file), then running the baseline, the baseline run fails to generate the baseline.csv, which later causes a failure when attempting to do a full simulation run.

Steps:
1.  Run JSON DUMP of the model
Model: https://github.com/PresencePG/grip-server/blob/develop/components/grip_sim_runner/gridlabd/simulation_models/virtual_islanding_ice/virtual_islanding_model_ice.glm
```
docker run --rm --name gridlabd_11_17_virtual_islanding_model_ice.glm_c687bc8f-69d7-4b48-a139-3a25994d51e3 -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3:/model -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py  -C  -W /model -i virtual_islanding_model_ice.glm -o virtual_islanding_model_ice_post_weather_pre_weather.json
```
2.  Convert the JSON dump to glm
```
docker run --rm --name gridlabd_11_17_virtual_islanding_model_ice_post_weather_post_weather_update.json_c687bc8f-69d7-4b48-a139-3a25994d51e3 -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3:/model -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u /usr/local/share/gridlabd/json2glm.py -i virtual_islanding_model_ice_post_weather_post_weather_update.json -o virtual_islanding_model_ice_post_weather.glm
```
3.  Run the Baseline

Commands:
```
docker run --rm --name gridlabd_11_17_virtual_islanding_model_ice_post_weather.glm_c687bc8f-69d7-4b48-a139-3a25994d51e3 -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3:/model -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py -B   -W /model -i virtual_islanding_model_ice_post_weather.glm -o virtual_islanding_model_ice_post_baseline.json
```

Full Output (Includes the -C run after the baseline):
```
UPDATING THE WEATHER FILE WITH: CA-Los_Angeles_Intl_Arpt.tmy3CREATING INITIAL JSON DUMP TO UPDATE THE WEATHER FILE+ CONTAINER_REMOVE_FLAG=--rm
+ SIMULATION_RUN_UUID=c687bc8f-69d7-4b48-a139-3a25994d51e3
+ SIMULATION_ID=11
+ SIMULATION_RUN_ID=17
+ TEMPDIR=/tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3
+ INPUT_MODEL_FILENAME=virtual_islanding_model_ice.glm
+ MODEL_OUTPUT_FILENAME=virtual_islanding_model_ice_post_weather_pre_weather.json
+ GRIDLABD_DOCKER_IMAGE=slacgrip/master:191022
+ GRIDLABD_COMPILE_ONLY=true
+ GRIDLABD_RUN_JSON2GLM=false
+ GRIDLABD_PYTHON_FILE=run_gridlabd_main.py
+ GRIDLABD_DASHB=false
+ GRIDLABD_CONTAINER_NAME=gridlabd_11_17_virtual_islanding_model_ice.glm_c687bc8f-69d7-4b48-a139-3a25994d51e3
+ cmd=
+ COMPILE_ONLY_FLAG=
+ DASH_B_FLAG=
+ RUN_JSON2GLM_FLAG=
+ '[' true == true ']'
+ '[' false == true ']'
+ COMPILE_ONLY_FLAG='-C '
+ mkdir -p /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3/tmp
+ '[' -z run_gridlabd_main.py ']'
+ echo 'Python event loop configured to run: run_gridlabd_main.py'
Python event loop configured to run: run_gridlabd_main.py
+ '[' false == true ']'
+ echo Running GridLAB-D Simulation with Python Event Loop
Running GridLAB-D Simulation with Python Event Loop
+ cmd='docker run --rm --name gridlabd_11_17_virtual_islanding_model_ice.glm_c687bc8f-69d7-4b48-a139-3a25994d51e3 -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3:/model -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py  -C  -W /model -i virtual_islanding_model_ice.glm -o virtual_islanding_model_ice_post_weather_pre_weather.json'
+ newline='
'
+ header='===================================================
Running GridLAB-D container
GridLAB-D Model=virtual_islanding_model_ice.glm
Simulation ID=11
Simulation Run ID=11
Simulation Run Temp directory=/tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3
Output Filename=virtual_islanding_model_ice_post_weather_pre_weather.json
GridLAB-D Docker Image=-w /model slacgrip/master:191022
GridLAB-D Container Name=gridlabd_11_17_virtual_islanding_model_ice.glm_c687bc8f-69d7-4b48-a139-3a25994d51e3
Python Event Loop: run_gridlabd_main.py
CONTAINER_REMOVE_FLAG:--rm
COMPILE_ONLY_FLAG=-C 

Run command: docker run --rm --name gridlabd_11_17_virtual_islanding_model_ice.glm_c687bc8f-69d7-4b48-a139-3a25994d51e3 -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3:/model -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py  -C  -W /model -i virtual_islanding_model_ice.glm -o virtual_islanding_model_ice_post_weather_pre_weather.json
===================================================
'
+ printf '===================================================
Running GridLAB-D container
GridLAB-D Model=virtual_islanding_model_ice.glm
[... 3786 lines cut ...]
docker.io/slacgrip/master:191022
+ set -e
+ docker run --rm --name gridlabd_11_17_virtual_islanding_model_ice_post_weather.glm_c687bc8f-69d7-4b48-a139-3a25994d51e3 -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3:/model -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py -C -W /model -i virtual_islanding_model_ice_post_weather.glm -o virtual_islanding_model_ice_post_weather_post_run.json
Setting path to folder:  /model
Compile Flag is Set
GridLAB-D 4.2.0-191022
Packetized module found, using virtual battery module for absorption.

No errors in file.
Warning(s) found in the model, printing gridlabd.wrn file contents...
WARNING  [INIT] : class_find_property(CLASS *oclass='solar', PROPERTYNAME name='Rated_kVA': property is deprecated
WARNING  [INIT] : last warning message was repeated 5 times
WARNING  [INIT] : loadshape_init(loadshape *ls={schedule->name='waterheater_demand_prob',...}) pulsed loadshapes are not validated
WARNING  [INIT] : last warning message was repeated 539 times
WARNING  [INIT] : Shunt capacitance of overhead line:line_node101A_load101A not calculated - invalid values
WARNING  [INIT] : last warning message was repeated 53 times
WARNING  [INIT] : Node:node_101A is parented to a swing node and will get folded into it.
WARNING  [INIT] : last warning message was repeated 26 times
WARNING  [INIT] : inv_supernode_103_battery: inverter:inv_supernode_103_battery is LOAD_FOLLOWING and/or power-factor regulating based on a meter or triplex_meter, ensure the inverter is connected inline with that object!
WARNING  [INIT] : last warning message was repeated 8 times
WARNING  [INIT] : class_find_property(CLASS *oclass='solar', PROPERTYNAME name='Rated_kVA': property is deprecated

+ echo Finished running docker run --rm --name gridlabd_11_17_virtual_islanding_model_ice_post_weather.glm_c687bc8f-69d7-4b48-a139-3a25994d51e3 -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3:/model -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py -C -W /model -i virtual_islanding_model_ice_post_weather.glm -o virtual_islanding_model_ice_post_weather_post_run.json
Finished running docker run --rm --name gridlabd_11_17_virtual_islanding_model_ice_post_weather.glm_c687bc8f-69d7-4b48-a139-3a25994d51e3 -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3:/model -v /tmp/grip/simulation_runs/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py -C -W /model -i virtual_islanding_model_ice_post_weather.glm -o virtual_islanding_model_ice_post_weather_post_run.json
```

jcald1, 2019-12-21:
[11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3.zip](https://github.com/dchassin/gridlabd/files/3990930/11-17__c687bc8f-69d7-4b48-a139-3a25994d51e3.zip)

dchassin, 2019-12-21:
@aivanova5, can you diagnose the root cause of this?

closed 2021-01-25 by aivanova5