arras-energy/gridlabd-old #506: Intermittent validation error on test_pole_NR.glm

aivanova5, 2020-01-07:
## Problem description
When validating develop there is an intermittent fail of autotest `powerflow/test_pole_NR.glm`

## Steps to reproduce
(Explain how to reproduce the behavior, e.g.,)
`gridlabd -W powerflow --validate`

## Expected behavior
100% pass rate

## System information
* GridLAB-D version: HiPAS GridLAB-D 4.2.1-200107 (develop_fix_websites) Darwin
* Operating system: `Darwin dhcp-swlan-public-84-172.slac.stanford.edu 19.2.0 Darwin Kernel Version 19.2.0: Sat Nov  9 03:47:04 PST 2019; root:xnu-6153.61.1~20/RELEASE_X86_64 x86_64`

## Supporting information
NA

## Attachments
NA

dchassin, 2021-05-17:
@aivanova5, when you find a simple reduced model that reliably causes the error, please attach it to this issue.

dchassin, 2021-05-17:
@ftuffner, FYI this is the issue we are using to track resolution of this bug.

dchassin, 2021-05-17:
The problem is likely in `module/powerflow/pole.cpp@pole::presync()` somewhere after line 270.

dchassin, 2021-05-17:
Initial workplace: 
1. Change link_fault_on/off to line->status = OPEN/CLOSED.
2. Make sure NR solver is flagged to redo admittance (e.g., NR_admittance_change=TRUE)
3. Check that fault_check handles all links, not just switches.

closed 2021-12-28 by dchassin