arras-energy/gridlabd-old #529: line_sensor properties throw an error in json2glm converted glm file (IEEE123 feeder model)

yeyanzhu, 2020-01-31:
## Problem description
Can not run json2glm converted glm file due to inability to initialize the line_sensor object
## Steps to reproduce
Take IEEE123 feeder model as the testing case, which contains the line sensor objects
1. using gridLAB-D to output json file from original feeder model file
2. using json2glm to convert json file to glm file
3. run power flow using the converted glm file
4. Initialization error will show, related to line-senor property "co-variance"

## Expected behavior
No error when running converted glm file

## System information
* GridLAB-D version: 
GridLAB-D 4.2.0-191209 (master) LINUX

## Supporting information
Error screenshot attached 
[image]

## Attachments
json file, converted glm file attached

[IEEE123Issue.zip](https://github.com/slacgismo/gridlabd/files/4137674/IEEE123Issue.zip)

dchassin, 2020-01-31:
Tested this JSON file with `develop` and found additional problems:
1. The timezone is not converted properly.
2. The conversion of `kmlhost` fails.

dchassin, 2020-01-31:
@yeyanzhu, could you post the original GLM file that generated the JSON file?

yeyanzhu, 2020-02-02:
I re-run the testing. The original GLM file/json file/converted glm file are all attached. 
I already manually removed the KMLhost part from the converted glm file.

[IEEE123_Testing.zip](https://github.com/slacgismo/gridlabd/files/4143429/IEEE123_Testing.zip)

aivanova5, 2020-02-03:
@yeyanzhu What are you trying to achieve by using line sensors? Line sensors were developed as a custom module for a specific project. In addition, line sensors are not widely used in the industry so there may be a better way to address what you're trying to do that avoids this problem.

Hitachi-ESL, 2020-02-04:
Hi Alyona, I am not using line sensors for any particular case study.  I just happened to use this model file for a general file conversation consistency testing.

aivanova5, 2021-01-26:
@yeyanzhu is this still an issue?

Hitachi-ESL, 2021-01-26:
Hello Alyona,

The bug is not an issue for me. Thanks for checking

Yanzhu

[-----WARNING-----] This email originated outside of Hitachi. [-----WARNING-----]
DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe.

closed 2021-01-26 by aivanova5