arras-energy/gridlabd-old #561: Collector unable to recognize groupid

aivanova5, 2020-03-10:
## Problem description
Can't use groupid to sort objects for a collector. A more concerning problem is that validation doesn't pick up the errors...

## Steps to reproduce
(Explain how to reproduce the behavior, e.g.,)
1. Go to the PR#559 `develop-docs-collector` that was set up and run the autotest in /tape/autotests/test_collector_triplex_meter.glm
2. Additionally go to /gldcore/autotest/test_groupid.glm and run is from validation folder 
NOTE: RUNNING VALIDATE DOESN'T PICK UP THE ERROR!?!?!?
~~~
dhcp-swlan-public-84-172:test_groupid alyona_slac$ gridlabd test_groupid.glm 
WARNING  [INIT] : house_e:0 has no parent triplex_meter defined; using static voltages
ERROR    [2000-01-01 00:00:00 PST] : aggregate group expression 'groupid=A' does not result in a set with a fixed class
ERROR    [2000-01-01 00:00:00 PST] : collector 22 'count(floor_area)' contains an aggregate that is not found in the group 'groupid=A'

ERROR    [2000-01-01 00:00:00 PST] : 2000-01-01 00:00:00 PST: object collector:22 stopped its clock (exec)!
ERROR    [2000-01-01 00:00:00 PST] : exec halted: synchronization failed
FATAL    [2000-01-01 00:00:00 PST] : shutdown after simulation stopped prematurely
FATAL    [2000-01-01 00:00:00 PST] : environment startup failed: No such file or directory
WARNING  [2000-01-01 00:00:00 PST] : last warning message was repeated 21 times
~~~

## Expected behavior
Reasonable output from collector

## System information
* GridLAB-D version: HiPAS GridLAB-D 4.2.1-200310 (develop_docs_collector) Darwin
* Operating system: Darwin dhcp-swlan-public-84-172.slac.stanford.edu 19.3.0 Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64 x86_64

## Supporting information
NA

## Attachments
NA

dchassin, 2020-03-11:
This only seems to happen when `groupid` is the first term in the criteria.  If the first term is something invariant, e.g., `class`, then this doesn't happen.

dchassin, 2020-03-11:
There are several problems:

1. Invariance is only based on the first criterion.  If any subsequent criteria are variant, the invariant violation is not detected. No suggested fix for this yet.

2. Groupid does not set it's invariance flag. Easy fix.

3. The aggregate test only check class, not all constants. Easy fix.

4. There is no global to override the invariance test if desired. Easy fix.

closed 2020-03-17 by dchassin