arras-energy/gridlabd-old #611: fail to use #on_exit command to catch non-zero exit-code

yeyanzhu, 2020-04-11:
## Problem description
Fail to use #on_exit command to catch no-zero exit-code when model compilation or model initialization fails.

## Steps to reproduce
1. Add the following commands to dump the error code to files for status checking.
__#on_exit 0 python3 ExitCheck.py ${exit_code}_
_#on_exit -1 python3 ExitCheck.py ${exit_code}_
The python code (ExitCheck.py) is to output exit_code to some file.

2. Use "_gridlabd -C main.glm_" command to run model compilation 
3. Here are some testings and results I got:
1)	When the model complication succeed, the command "_#on_exit 0_ " will be triggered and exit-code is recorded as zero in the status recording file. However, when exit_code is non-zero, the _"#on_exit -1"_ command fails to work correctly, nothing is recorded. Here, the weather file (CA-Bakersfield_Meadows_Field.tmy3) is removed to create model compilation error. 

 2)	Then I change the second command to _#on_exit 5 python3 ExitCheck.py ${exit_code}_, since in json file the exit_code is recorded as 5. Then, the command of "#on_exit 5" will be triggered to output exit_code to the status recording file. However, in the status recording file, the exit_code is recorded as zero instead of non-zero value, e.g. 5.

## Expected behavior
Use the command  "_#on_exit -1 python3 ExitCheck.py ${exit_code}_" to catch model failure status correctly.

## System information
* GridLAB-D version: 
HiPAS GridLAB-D 4.2.1-200223

## Supporting information

## Attachments
The attachment include the IEEE123 feeder model file, python file, and status file.
[OnExit_Testingfile.zip](https://github.com/slacgismo/gridlabd/files/4464224/OnExit_Testingfile.zip)

dchassin, 2020-04-20:
Note, this problem currently only happens on the docker image. It does not occur on macOS.

dchassin, 2020-04-20:
The problem does not occur when optimization is turned off.

closed 2020-04-24 by dchassin