arras-energy/gridlabd-old #672: glm2glm converter failed to record schedule information

yeyanzhu, 2020-06-02:
## Problem description
When using gridlabd to merge multiple glm files onto one single glm file, the load schedule information is not written onto the merged glm file.

## Steps to reproduce
1. Set the following global file-save parameters:
`#set glm_save_options=ORIGINAL`
`#set filesave_options=ALLINITIAL`
2. Using the following command to convert onto one single glm file
`gridlabd BU0001_glmfile_ts.glm -o merged.glm`
3. The schedules of load_shape_A, load_shape_B,... are not written onto merged.glm

## Expected behavior
The schedule information should be recorded in merged glm file.

## System information
* GridLAB-D version: 
HiPAS GridLAB-D 4.2.1-200527 (develop) Linux
(using GridLAB-D docker)

## Attachments
The glm model file and output file
[BU0001.zip](https://github.com/slacgismo/gridlabd/files/4715735/BU0001.zip)

yeyanzhu, 2020-06-09:
A simple model (4-node system) is uploaded to demonstrate the issue. 
[IEEE-4node.zip](https://github.com/slacgismo/gridlabd/files/4755360/IEEE-4node.zip)

dchassin, 2020-06-12:
Please use the option `#set glm_save_options=LEGACY` to include the schedules.

closed 2020-06-22 by aivanova5