arras-energy/gridlabd-old #675: json2glm converter fails to include schedule information onto converted glm file

yeyanzhu, 2020-06-03:
## Problem description
When gridlabd is used to convert json file to glm file, the schedule information listed in JSON file fails to be included in glm output file. The following command is used for file conversion:
     `gridlabd -C res.json -o test.glm`
In the json file, the filesave options is set as:
     `glm_save_options : ORIGINAL`
    `filesave_options: ALLINITIAL`

## Steps to reproduce
1. Run the gridlabd command in the model folder
      `gridlabd -C res.json -o test.glm`
2. In the converted glm file, the schedule information `load_shape_A,load_shape_B,load_shape_C` is not included.

## System information
* GridLAB-D version: 
     `HiPAS GridLAB-D 4.2.1-200527 (develop) Linux`
(run using docker image)

## Attachments
Model file, json file and converted glm file output
[BU0001.zip](https://github.com/slacgismo/gridlabd/files/4725835/BU0001.zip)

yeyanzhu, 2020-06-10:
A simple model (IEEE 4 node system) is uploaded to demonstrate the issue. Gridlabd command used:
`gridlabd -C 4noderes.json -o converted.glm`
Issues:
- The load-shape schedule information of object `load4` is not included in `converted.glm` file
- The converted object `load4` in `converted.glm` file do not keep the original property settings. 
[IEEE4node_json2glm.zip](https://github.com/slacgismo/gridlabd/files/4755466/IEEE4node_json2glm.zip)

dchassin, 2020-06-12:
The uploaded file `res.json` does not have `glm_save_options` = `ORIGINAL` as stated in the problem description. The global is not set at all. If I add the following to the `"globals"` dict in `res.json`, then the result is as expected:

~~~
"glm_save_options" : {
	"type" : "enumeration",
	"access" : "PUBLIC",
	"value" : "LEGACY"
}
~~~

Recommend correcting the program the write the JSON file.

closed 2020-06-12 by dchassin