arras-energy/gridlabd-old #701: Weather subcommand doesn't work from python shell

aivanova5, 2020-06-24:
## Problem description
When running a glm file that uses `#weather get <filename.tmy3>` an error occurs. Works correctly when running `gridlabd test.glm` 
~~~
alyona_slac$ python3 main.py
ERROR    [INIT] : GldMain::subcommand(format='%s/gridlabd-%s',...): command 'gridlabd/gridlabd-weather get CA-Chino_Airport.tmy3
' returns code 127
main.glm(32): #weather macro is not recognized
FATAL    [INIT] : shutdown after command line rejected
Model profiler results
======================

Class            Time (s) Time (%) msec/obj
---------------- -------- -------- --------
================ ======== ======== ========
Total              0.000    100.0%      nan
~~~

## Steps to reproduce
(Explain how to reproduce the behavior, e.g.,)
Run `python3 test.py` in attached 

## Expected behavior
Download of the weather file to be used within the model

## System information
* GridLAB-D version: HiPAS GridLAB-D 4.2.1-200623 (develop) Darwin;
* Operating system:  Docker exhibits this issue but freezes indefinitely; Darwin PC95555 19.5.0 Darwin Kernel Version 19.5.0: Tue May 26 20:41:44 PDT 2020; root:xnu-6153.121.2~2/RELEASE_X86_64 x86_64

## Supporting information
(Add any other context about the problem here.)

## Attachments
(If applicable, attach screenshots, supporting files, and other data need to help explain and reproduce the problem.)

aivanova5, 2020-07-02:
I've diagnosed the issue with the TX file. See the first line as follows: 

~~~
722670,LUBBOCK INTERNATIONAL AP,TX,-6,33.667,-101.817,992,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
~~~
 The additional commas are corrupting the simulation.

dchassin, 2020-07-03:
Ok, we’ll need to make the reader more robust to extra columns.

D

dchassin, 2021-04-27:
This problem is not longer an issues.  The following works fine:
~~~
#weather get "TX-Lubbock_International_Ap.tmy3"
#input "TX-Lubbock_International_Ap.tmy3"
~~~

closed 2021-04-27 by dchassin