arras-energy/gridlabd-old #708: Segmentation Fault on GRIP Recovery Simulation When Using the TX-Lubbock_Intl_Ap_722670TYA.tmy3 Weather File

jcald1, 2020-07-02:
## Problem description
When running a simulation from a model that was JSON-dumped then JSON-to-GLM-converted using the TX-Lubbock_Intl_Ap_722670TYA.tmy3 weather file, we receive a segmentation fault error.  We only periodically get the segmentation fault error when using the CA-San_Francisco_Intl_Ap.tmy3 weather file.

## Steps to reproduce
1.  Extract the attached zip file to `/tmp/grip/simulation_runs/5-8__6fcb59eb-3145-4fbf-b371-575f3ad8b734`.  Don't extract to the user home directory.
1.  Run `cd /tmp/grip/simulation_runs/5-8__6fcb59eb-3145-4fbf-b371-575f3ad8b734`
1.  Run `docker run --rm --name gridlabd_5_8_recovery_ieee37_output.glm_6fcb59eb-3145-4fbf-b371-575f3ad8b734 -v /tmp/grip/simulation_runs/5-8__6fcb59eb-3145-4fbf-b371-575f3ad8b734:/model -v /tmp/grip/simulation_runs/5-8__6fcb59eb-3145-4fbf-b371-575f3ad8b734/tmp:/tmp -v /var/run/docker.sock:/var/run/docker.sock -w /model slacgrip/master:191022 python3 -u run_gridlabd_main.py -W /model -i recovery_ieee37_output.glm -o recovery_ieee37_output_post_run.json`
1.  The process exits with code `139` (segmentation fault).

## Expected behavior
The process should exit with code `0`.

## System information
* GridLAB-D version: GridLAB-D 4.2.0-191022 (master) LINUX
* Operating system: (Docker Container): Linux 35905cd1ca48 5.4.11-050411-generic #202001120738 SMP Sun Jan 12 12:40:39 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

## Supporting information

## Attachments
Zip file attached
[segmentation_fault.zip](https://github.com/slacgismo/gridlabd/files/4866831/segmentation_fault.zip)

aivanova5, 2020-07-02:
The steps to reproduce can be reduced to `gridlabd recovery_ieee37_output.glm`

aivanova5, 2020-07-02:
I've diagnosed the issue with the TX file. See the first line as follows: 

~~~
722670,LUBBOCK INTERNATIONAL AP,TX,-6,33.667,-101.817,992,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
~~~
 The additional commas are corrupting the simulation.

closed 2020-07-02 by aivanova5