arras-energy/gridlabd-old #738: Segmentation error occurred in the Ica_internal.py run

HectorTR, 2020-09-02:
Segmentation error after the simulation ends. 

Information from terminal module:

Missing separate debuginfos, use: yum debuginfo-install bzip2-libs-1.0.6-26.el8.x86_64 glibc-2.28-101.el8.x86_64 keyutils-libs-1.5.10-6.el8.x86_64 krb5-libs-1.17-18.el8.x86_64 libcom_err-1.45.4-3.el8.x86_64 libcurl-minimal-7.61.1-12.el8.x86_64 libgcc-8.3.1-5.el8.0.2.x86_64 libnghttp2-1.33.0-1.el8_0.1.x86_64 libselinux-2.9-3.el8.x86_64 libstdc++-8.3.1-5.el8.0.2.x86_64 libxcrypt-4.1.1-4.el8.x86_64 ncurses-libs-6.1-7.20180224.el8.x86_64 openssl-libs-1.1.1c-15.el8.x86_64 xz-libs-5.2.4-3.el8.x86_64 zlib-1.2.11-13.el8.x86_64
--Type <RET> for more, q to quit, c to continue without paging-- 

Thread 1 "gridlabd.bin" received signal SIGSEGV, Segmentation fault.
0x00007fa740e2db88 in _PyDict_Next (phash=0x0, pvalue=0x7fffbe2adcd8, pkey=0x7fffbe2adcd0, ppos=0x7fffbe2adcc8, op=<optimized out>) at Objects/dictobject.c:1680
1680            while (i < n && entry_ptr->me_value == NULL) {
(gdb) list 
1675        else {
1676            Py_ssize_t n = mp->ma_keys->dk_nentries;
1677            if (i < 0 || i >= n)
1678                return 0;
1679            entry_ptr = &DK_ENTRIES(mp->ma_keys)[i];
1680            while (i < n && entry_ptr->me_value == NULL) {
1681                entry_ptr++;
1682                i++;
1683            }
1684            if (i >= n)
(gdb) where
#0  0x00007fa740e2db88 in _PyDict_Next (phash=0x0, pvalue=0x7fffbe2adcd8, pkey=0x7fffbe2adcd0, ppos=0x7fffbe2adcc8, op=<optimized out>) at Objects/dictobject.c:1680
#1  _PyDict_Next (op=<optimized out>, ppos=0x7fffbe2adcc8, pkey=0x7fffbe2adcd0, pvalue=0x7fffbe2adcd8, phash=0x0) at Objects/dictobject.c:1655
#2  0x00007fa740e3a401 in _PyModule_ClearDict (d=0x7fa74145da00) at Objects/moduleobject.c:614
#3  0x00007fa740e3a70e in _PyModule_Clear (m=m@entry=0x7fa741454e90) at Objects/moduleobject.c:577
#4  0x00007fa740ee88ed in PyImport_Cleanup () at Python/import.c:558
#5  0x00007fa740ef630f in Py_FinalizeEx () at Python/pylifecycle.c:1195
#6  Py_FinalizeEx () at Python/pylifecycle.c:1120
#7  0x000000000048c4a9 in python_embed_term () at gldcore/python_embed.cpp:42
#8  0x00000000004791d0 in GldMain::~GldMain (this=0x19b0550, __in_chrg=<optimized out>) at gldcore/main.cpp:193
#9  0x0000000000420ee0 in main (argc=<optimized out>, argv=0x7fffbe2adf18) at gldcore/main.cpp:78
(gdb)

aivanova5, 2020-09-08:
Can you add the info on location of the file, the commands used to run your code and OS info?

HectorTR, 2020-09-09:
Hi Alyona, I used gridlabd ica_ieee123.glm to run the code.
macOS Catalina version 10.15.6

closed 2021-01-26 by aivanova5