arras-energy/gridlabd-old #796: Docker hangs when converting csv

saraborc, 2020-11-05:
## Problem description

When IEEE123 is run using docker, the process hangs when using the csv converter `#input "ica_config.csv" -f config -t config`. This problem only occurs when running the model using docker. 

## Steps to reproduce

1. Go to branch `use_case/integration_capacity_test`
2. Make sure that `ieee123/ica_ieee123.glm` has line `#input "ica_config.csv" -f config -t config` uncommented.
2. From repo `gridlabd-models` run: `docker run -itv "$(pwd)":/test slacgismo/gridlabd:latest` 
2. `cd` into `ICA_models`
4. Run: `python3 ica_main.py`

## Expected behavior

`ica_config.csv` will not be converted, and process will hang.

## System information

* GridLAB-D version: HiPAS GridLAB-D 4.2.12-201009 (master) Linux
* Operating system: ( x86_64 GNU/Linux

dchassin, 2020-11-05:
Does this happen with `beau harbors-13`?

saraborc, 2020-11-05:
Not sure I understand. What is `beauharnois-13`?

dchassin, 2020-11-05:
The version info is patch 12 to version 4.2 (which is named `beauharnois`).  Have you pulled the latest docker image? Or is this the latest image you're getting when you pull the latest image?

dchassin, 2020-11-05:
Also, the branch `use_case/integration_capacity_test` is not available on Github.  Have you pushed it yet?

saraborc, 2020-11-05:
Yes, I've pulled the latest docker image. I am using tag 'latest'. The image ID is `0a08f9ff2f7c`. And yes, I've pushed! It's on `gridlabd-models`

dchassin, 2020-11-05:
Interesting. That's the right id, but when I run
`docker run -it slacgismo/gridlabd:latest gridlabd --version=all` I get the following output for the version info:
```
HiPAS GridLAB-D 4.2.13-201027 (master) Linux
```
which is not what you reported above.

dchassin, 2020-11-05:
One thing I see is that your print statements are being buffered and are not being displayed right away.  You can cause them to be output right away using `print(...,flush=True)`.

dchassin, 2020-11-05:
I think it's waiting on keyboard input. It starts when I hit Ctrl-D whenever it pauses.  Somewhere in your code is something that's reading /dev/stdin. A quick fix is to use the command `python3 ica_main.py < /dev/stdin`.  But ultimately, you're going to have to find what's reading the input and remove it.

dchassin, 2020-11-05:
BTW, did you intend to run a realtime simulation?

saraborc, 2020-11-05:
That quick fix worked for me! I intended to run the ICA analysis on IEEE123.

dchassin, 2020-11-05:
IEEE-123 runs a realtime simulation when it is not configured to use a start and stop date. Take a look at the config you commented out and consider adding the items you find listed there in your new config.

aivanova5, 2020-11-05:
I tested it on both MAC and Docker and the issue only arises on Docker.

aivanova5, 2020-11-05:
Here's the barebones test file. Note, only apparent on Docker.

[Archive.zip](https://github.com/slacgismo/gridlabd/files/5498023/Archive.zip)

dchassin, 2020-12-04:
@aivanova5, I am unable to reproduce the after install python@3.9.  Can you confirm that this problem is fixed when upgrading?  If so, we could simply change the required version in configure.ac to ensure that 3.9 is used.

closed 2020-12-11 by dchassin