arras-energy/gridlabd-old #816: on_term doesn't pipe correctly

aivanova5, 2020-12-05:
## Problem description

Task: check a validation autotest output to contain a string within gridlabd.err or gridlabd.wrn within an embedded model the files mentioned below, where the validation should raise an error if WARNING [check] is present in gridlabd.err

Using `on_exit` macro doesn't seem to correctly catch the fact that the embedded model `check_model.glm` generates a warning by using grep

PR: 

gridlabd -W gldcore/scripts --validate
More specifically: `test_subcommand_check.glm`

## Steps to reproduce

See above 

## Expected behavior

See above

## System information

* GridLAB-D version: HiPAS GridLAB-D 4.2.14-201204 (develop-add-check-subcommand) Darwin
* Operating system: Darwin pc95555.lan 19.6.0 Darwin Kernel Version 19.6.0: Thu Oct 29 22:56:45 PDT 2020; root:xnu-6153.141.2.2~1/RELEASE_X86_64 x86_64

## Supporting information

(Add any other context about the problem here.)

## Attachments

(If applicable, attach screenshots, supporting files, and other data need to help explain and reproduce the problem. If you encountered a program fault or internal error, and you have built GridLAB-D with `-O0 -g`, then on Linux systems you can attach the output of the following command:)
~~~
bash$ ulimit -c unlimited
bash$ gridlabd <your command options...>
... crash dumps core ...
bash$ gdb -batch -ex "backtrace" $(gridlabd --version=install) ./core
~~~

(If you are running on macOS, you must use `lldb` to do the following.)
~~~
bash$ ulimit -c unlimited
bash$ ls -lt /cores # find the most recent core dump
bash$ lldb -c /cores/core.<number>
(lldb) bt -- backtrace
~~~

aivanova5, 2020-12-05:
`on_term` from python also doesn't handle the error for validation
~~~
#check ../check_model.glm 
module validate_check_model;
~~~

~~~
#print "model loaded"
module powerflow;
module climate;

clock {
	timezone US/CA/Los Angeles;
	starttime '2000-01-01 00:00:00';
	stoptime '2001-01-01 00:00:00';
}

#weather get CA-Chino_Airport.tmy3
object climate 
{
	name weather;
	tmyfile "CA-Chino_Airport.tmy3";
}

object node { 
      name main;
      phases "ABCN";
     // bustype SWING;
      nominal_voltage 2401.7771;
 }
~~~

~~~validate_check_model.py which gets installed from python_extras into /usr/local/share/gridlabd folder
import sys
def on_term(t) : 
	with open('gridlabd.err') as gld_err : 
		assert("WARNING [check]" not in gld_err.read()  )	

	with open('gridlabd.wrn') as gld_wrn : 
		assert("WARNING [check]" not in gld_wrn.read()  )	
~~~

OUTPUT: 
pc95555:oct262020 alyona_slac$ gridlabd -W gldcore/scripts --validate
Starting validation test in directory '/Users/alyona_slac/gridlabd/oct262020/gldcore/scripts'
Traceback (most recent call last):_subcommand_check.glm...
  File "/usr/local/opt/gridlabd/4.2.14-201204-develop-add-check-subcommand/share/gridlabd/validate_check_model.py", line 4, in on_term
    assert("WARNING [check]" not in gld_err.read()  )	
AssertionError

Validation report:
2 models tested
2 tests succeeded
100% success rate
Total validation elapsed time: 9.1 seconds

dchassin, 2021-03-31:
@aivanova5, can you determine whether this problem still exists after the pipe upgrade.

closed 2023-09-27 by dchassin