arras-energy/gridlabd-old #826: csv converter only works with certain csv files

saraborc, 2020-12-17:
## Background
The csv converter works when the csv is saved as Macintosh Comma Separated (.csv), but encounters errors when used with other types. 

For example, when the `input_config.csv` for the ICA script was saved as CSV UTF-8 (the default csv on Mac Excel), blanks in `input_config.csv` raise this error:

`UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc8 in position 0: invalid continuation byte`

## Proposed solution
We can modify the csv converter to be compatible with any csv, or raise a warning when the file is not in the expected format.

## Additional context
To replicate the error on the ICA script, navigate to the branch `use_case/integration_capacity_test` on the `gridlabd-models` repo. Save `ica_config.csv` as CSV UTF-8 and run:
`gridlabd ica_ieee123.glm`

The following error should be raised: `UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc8 in position 0: invalid continuation byte`

To replicate on a simplified script, use the attached zip file. First run:
`%host gridlabd conv_test.glm
`
Then, replace the `input` command in `conv_test.glm` with:
`#input "conv_test_config_utf.csv" -f config -t config
`
This uses the csv converter on a CSV UTF-8 file. Compare the results.

[convtest.zip](https://github.com/slacgismo/gridlabd/files/5711895/convtest.zip)

dchassin, 2020-12-17:
Does it make sense to instruct users only use "Comma Separated Values (CSV)" and not use any other format?  Or would you prefer adding an option to specify the encoding?

saraborc, 2020-12-17:
Yes, I think that instruction would be a good, quick fix! I think it's fine that the converter only works with certain csv formats, but we could save the user some time/confusion by raising a warning if their config file is in the wrong format.

dchassin, 2020-12-17:
I see your point.  This is a broader issue with gridlabd calls to python.  The exceptions are reported verbatim and sometimes the underlying problem is obscured by the many layers of code between the top level request and the ultimate demise.  IThe problem is that gridlabd does not dump the entire traceback of the python exception.  

I'm curious to know whether enabling `verbose` or `debug` changes what you see.