arras-energy/gridlabd-old #839: Single file dump, dumps properties that prevent the model from running

aivanova5, 2021-02-04:
## Problem description

For example, using `glm_save_options=MINIMAL` with `-o` on IEEE123 with explicit houses turned out results in the single file containing properties that can't run with error as below

~~~
property load of house_1 (house:316) could not be set to value '+0+0j'
~~~

The model properties commented out below all exhibit this error: 

~~~
object residential.house {
	parent "meter_343";
	name "house_343";
	guid "6F1328DE921CC7031A6E48375DC8463B";
	weather "weather";
	floor_area "+1578.19 sf";
	gross_wall_area "+1297.46 sf";
	ceiling_height "+8 ft";
	aspect_ratio "+1.5";
	envelope_UA "+285.352 Btu/degF/h";
	window_wall_ratio "+0.15";
	number_of_doors "+4";
	exterior_wall_fraction "+1";
	interior_exterior_wall_ratio "+1.5";
	exterior_ceiling_fraction "+1";
	exterior_floor_fraction "+1";
	window_shading "+0.67";
	window_exterior_transmission_coefficient "+0.6";
	airchange_per_hour "+0.5 unit/h";
	airchange_UA "+111.45 Btu/degF/h";
	UA "+396.801 Btu/degF/h";
	thermostat_deadband "+2 degF";
	thermostat_cycle_time "120";
	heating_setpoint "+70 degF";
	cooling_setpoint "+75 degF";
	design_heating_setpoint "+70 degF";
	design_cooling_setpoint "+75 degF";
	design_heating_capacity "+48000 Btu/h";
	design_cooling_capacity "+48000 Btu/h";
	cooling_design_temperature "+107.962 degF";
	heating_design_temperature "+30.922 degF";
	design_peak_solar "+195 Btu/h/sf";
	design_internal_gains "+4330.48 Btu/h";
	mass_solar_gain_fraction "+0.5 pu";
	mass_internal_gain_fraction "+0.5 pu";
	auxiliary_heat_capacity "+20000 Btu/h";
	aux_heat_deadband "+2 degF";
	aux_heat_temperature_lockout "+10 degF";
	aux_heat_time_delay "+300 s";
	duct_pressure_drop "+0.5 inH2O";
	fan_design_power "+317.77 W";
	fan_design_airflow "+1394.27 cfm";
	heating_COP "+2.5 pu";
	cooling_COP "+3.5 Btu/kWh";
	air_temperature "+74.0402 degF";
	outdoor_temperature "+74 degF";
	mass_heat_capacity "+2710.57 Btu/degF";
	mass_heat_coeff "+6641.85 Btu/degF/h";
	mass_temperature "+74.0402 degF";
	air_volume "+12625.5 cf";
	air_mass "+927.973 lb";
	air_heat_capacity "+668.698 Btu/degF";
	latent_load_fraction "+0.3 pu";
	total_thermal_mass_per_floor_area "+2 Btu/degF/sf";
	interior_surface_heat_transfer_coeff "+1.46 Btu/h/degF/sf";
	number_of_stories "+1";
	auxiliary_strategy "DEADBAND";
	system_mode "OFF";
	last_system_mode "OFF";
	cooling_system_type "HEAT_PUMP";
	auxiliary_system_type "ELECTRIC";
	fan_type "ONE_SPEED";
	last_mode_timer "21600";
	Rroof "+30 degF*sf*h/Btu";
	Rwall "+19 degF*sf*h/Btu";
	Rfloor "+22 degF*sf*h/Btu";
	Rwindows "+2.12766 degF*sf*h/Btu";
	Rdoors "+5 degF*sf*h/Btu";
	hvac_breaker_rating "+200 A";
	//panel "+0+0j";
	panel.energy "+0+0i kVAh";
	panel.power "+0+0i kVA";
	panel.peak_demand "+0+0i kVA";
	panel.heatgain "+0 Btu/h";
	panel.cumulative_heatgain "+0 Btu";
	panel.heatgain_fraction "+0 pu";
	panel.current_fraction "+0 pu";
	panel.impedance_fraction "+0 pu";
	panel.power_fraction "+0 pu";
	panel.power_factor "+0";
	panel.constant_power "+0+0i kVA";
	panel.constant_current "+0+0i kVA";
	panel.constant_admittance "+0+0i kVA";
	panel.voltage_factor "+0 pu";
	panel.breaker_amps "+0 A";
	panel.configuration "IS110";
	// class.parent = residential.residential_enduse
	//shape "+0";
	//load "+0+0j";
	energy "+0+0i kVAh";
	power "+0+0j kVA";
	peak_demand "+0+0i kVA";
	heatgain "+0 Btu/h";
	cumulative_heatgain "+0 Btu";
	heatgain_fraction "+0 pu";
	current_fraction "+0 pu";
	impedance_fraction "+0.2 pu";
	power_fraction "+0.8 pu";
	power_factor "+0.97";
	constant_power "+0+0j kVA";
	constant_current "+0+0j kVA";
	constant_admittance "+0+0j kVA";
	voltage_factor "+0 pu";
	breaker_amps "+200 A";
	configuration "IS220";
}
~~~

## Steps to reproduce

(Explain how to reproduce the behavior, e.g.,)

`gridlabd -D glm_save_options=MINIMAL model/ieee123.glm -o ieee123_single_file.glm`

## Expected behavior

Functional run

## System information

* GridLAB-D version: HiPAS GridLAB-D 4.2.16-210124 (develop-complex-formatting-fix) Darwin
* Operating system: Darwin pc95555.lan 19.6.0 Darwin Kernel Version 19.6.0: Tue Nov 10 00:10:30 PST 2020; root:xnu-6153.141.10~1/RELEASE_X86_64 x86_64

## Supporting information

n/a

## Attachments

n/a

dchassin, 2021-03-31:
Should `INITIAL` be used to create a loadable model?

dchassin, 2021-12-28:
This problem no longer occurs with `HiPAS GridLAB-D 4.2.29-211225 (develop) Darwin`

closed 2021-12-28 by dchassin