arras-energy/gridlabd-old #841: JSON output data changes due to gridlabd version change

yeyanzhu, 2021-02-09:
## Problem description
The json output changes caused by Gridlabd version change

## Steps to reproduce
1. Run the following command using different versions of Gridlabd docker image
`gridlabd -D filesave_options=ALLINITIAL -C IEEE_4_node.glm -o test.json`
Old version: slacgismo/gridlabd:beauharnois-08
Latest version: slacgismo/gridlabd:latest
2. Compare two json output file "test_olddocker.json", "test_newdocker.json"
Here are the differences:
By using the old version docker image, the json data includes the complete property list of each object. 
By using the latest version docker image, some properties are missing, e.g. switch:status, switch:phase_A_status, node:bustype, etc.  

## Expected behavior
Have a complete property list of each object in JSON output

## System information
* GridLAB-D version latest: 
HiPAS GridLAB-D 4.2.15-210114 (master) Linux
* GridLAB-D version older: 
HiPAS GridLAB-D 4.2.1-200522 (master) Linux

## Attachments
glm model file, json output file

[filesave.zip](https://github.com/slacgismo/gridlabd/files/5954091/filesave.zip)

dchassin, 2021-02-11:
@yeyanzhu, this is normal.  Default property values are not output in JSON anymore when the `MINIMAL` flag is set on `filesave_options`. If you change `ALLINITIAL` to `ALL|INITIAL` that will eliminate the implicit `MINIMAL` value in `ALLINITIAL` and give you the `ALL` and `INITIAL` behavior you want.

In the meantime, we are going to fix the documentation of `filesave_options` to better explain this subtlety.

yeyanzhu, 2021-02-11:
@dchassin when I change the setting to ALL|INITIAL as follows:
`gridlabd -D filesave_options=ALL|INITIAL -C IEEE_4_node.glm -o test.json`
It will fail with the following error message:
`bash: INITIAL: command not found`
I tried other possible combination, e.g. ALL|MINIMAL, it also fails.

dchassin, 2021-02-11:
When on the command line you’ll have to enclosed it single quotes to prevent the shell from parsing it as a pipe, i.e., `gridlabd -D filesave_options='ALL|INITIAL' -C IEEE_4_node.glm -o test.json`

yeyanzhu, 2021-02-11:
@dchassin Thanks, It works as expected after adding single quotes.

closed 2021-03-31 by dchassin