arras-energy/gridlabd-old #900: Not all glyphs are displayed in folium HTML map viewer converter

avpeery, 2021-05-01:
## Problem description

When displaying folium HTML map view - some glyphs (glyphs for poles) do not show.  The problems differ depending on which --glyphs=<choice> you select. Valid choices are fa and glyphicons.

## Steps to reproduce

N/A

## Expected behavior

Excepted to see cross glyph where there are poles on map viewer.

dchassin, 2021-05-11:
What are the chances we could modify folium so we could make our own icons?

avpeery, 2021-05-11:
It is possible to fork folium to modify it to allow for v5 fontawesome icons. I think it's also possible to do custom icons with a src url. This source is possibly dated though from 2015 on custom icons: https://ocefpaf.github.io/python4oceanographers/blog/2015/11/02/icons/

closed 2021-06-08 by dchassin