arras-energy/gridlabd-old #933: KVA Unit does not work in property setting of Commercial Parking module (Commercial)

yeyanzhu, 2021-06-24:
## Problem description
The KVA unit in property setting in Commercial Parking module fail to set the value in KVA. The value works in VA.

## Steps to reproduce
Using the attached GLM to run simulation, the measured power value in meter recorder file is not consistent with defined parking object.  The screenshot of recorder file is attached

## System information
* GridLAB-D version (docker image)
HiPAS GridLAB-D 4.2.17-210226 (master) Linux

[TestParkingl.zip](https://github.com/slacgismo/gridlabd/files/6712244/TestParkingl.zip)

dchassin, 2021-06-24:
Be aware the in GridLAB-D units are case sensitive.  Be sure you the units *exactly* as in `gldcore/unitfile.txt`.  In this case, it's possible you are using `KVA`, when the unit is `kVA`.

dchassin, 2021-06-24:
For now, please omit the units from the models until we can correctly document the units used in the model.

dchassin, 2021-06-24:
The internal units are `W` or `VA`.  When you omit the units, you must assume these are the units used.

closed 2021-07-13 by dchassin