arras-energy/gridlabd-old #950: Segmentation error in Python interaction

mlamlamla, 2021-07-21:
## Problem description

I start a glm model (gridlabd model.glm) which imports a Python-based gridlabd_functions module with an on_precommit() function to be executed at each event. The model fails after start up with a segmentation error:
 
[root@dadbfec6436d docker_TESS_dep]# gridlabd model_woPV.glm
WARNING  [INIT] : class_find_property(CLASS *oclass='solar', PROPERTYNAME name='phases': property is deprecated
WARNING  [INIT] : last warning message was repeated 4 times
WARNING  [INIT] : house:27: EIA 2015 implicit enduse data is experimental
/usr/local/bin/gridlabd: line 366:  1115 Segmentation fault      "$bindir/gridlabd.bin" "$@"

## Steps to reproduce

- I use the gridlabd docker image slacgismo/gridlabd from July 16, 2021 (54ad414a6944).
-  The code for testing is stored in the slacgismo/TESS repo, in folder simulation, on branch market_simulation_updates : https://github.com/slacgismo/TESS/tree/market_simulation_updates/simulation
- Access the docker container (slacgismo/gridlabd) and refer to the simulation folder of the TESS repo : docker run -it -v ~/Documents/path-to-TESS/simulation:/docker_TESS_dep slacgismo/gridlabd:latest
- From within there, run: gridlabd model_woPV.glm .
- Check the module gridlabd_fcts_test imported by model_woPV.
- The test throws segmentations errors:
--- If I import the module at the beginning of the module gridlabd_fcts_test.
--- It does not do so, when I exclude the solar module from the glm file.
--- It does not do so, if I import requests within the on_precommit().

## Expected behavior

/usr/local/bin/gridlabd: line 366:  1115 Segmentation fault      "$bindir/gridlabd.bin" "$@"
--> The simulation stops.

## System information

* GridLAB-D version: HiPAS GridLAB-D 4.2.24-210716 (master) Linux
* Operating system: Linux dadbfec6436d 5.10.25-linuxkit #1 SMP Tue Mar 23 09:27:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

dchassin, 2021-07-21:
Confirmed that there is a rogue pointer.  The `shape` value of the implicit end-uses is getting overwritten at some point after `init()` and before `sync()`:
~~~
Program received signal SIGSEGV, Segmentation fault.
0x00000000005f6ed4 in enduse_sync (e=e@entry=0x3823fc0, pass=pass@entry=PC_PRETOPDOWN, t1=t1@entry=1626880826) at gldcore/enduse.cpp:249
249		TIMESTAMP rt = (e->shape && e->shape->type != MT_UNKNOWN) ? e->shape->t2 : TS_NEVER;
(gbd) p *e
$1 = {total = {r = 0, i = 0, f = J}, energy = {r = 0, i = 0, f = J}, demand = {r = 0, i = 0, f = J}, config = 0, breaker_amps = 0, 
  admittance = {r = 0, i = 0, f = J}, current = {r = 0, i = 0, f = J}, power = {r = 0, i = 0, f = J}, motor = {{power = {r = 0, i = 0, 
        f = J}, impedance = {r = 0, i = 0, f = J}, inertia = 0, v_stall = 0, v_start = 0, v_trip = 0, t_trip = 0}, {power = {r = 0, i = 0, 
        f = J}, impedance = {r = 0, i = 0, f = J}, inertia = 0, v_stall = 0, v_start = 0, v_trip = 0, t_trip = 0}, {power = {r = 0, i = 0, 
        f = J}, impedance = {r = 0, i = 0, f = J}, inertia = 0, v_stall = 0, v_start = 0, v_trip = 0, t_trip = 0}, {power = {r = 0, i = 0, 
        f = J}, impedance = {r = 0, i = 0, f = J}, inertia = 0, v_stall = 0, v_start = 0, v_trip = 0, t_trip = 0}}, electronic = {{power = {
        r = 0, i = 0, f = J}, inertia = 0, v_trip = 0, v_start = 0}, {power = {r = 0, i = 0, f = J}, inertia = 0, v_trip = 0, 
      v_start = 0}}, impedance_fraction = 0.80000000000000004, current_fraction = 0, power_fraction = 0.20000000000000001, 
  power_factor = 0.97999999999999998, voltage_factor = 0, heatgain = 0, cumulative_heatgain = 0, heatgain_fraction = 1, gas_fraction = 0, 
  name = 0x7fec33a39a1c "DISHWASHER", shape = 0x4052000000000000, t_last = 1626880826, end_obj = 0x0, next = 0x38134b0}
~~~

dchassin, 2021-07-21:
The rogue pointer usage comes during the load of the Python `requests` module.  Upgrading `requests` does not fix the problem.

The problem does not occur if `requests` is imported in the `precommit()` function instead.  This is obviously not ideal because the rogue pointer is still present, but it's just hitting somewhere else in memory. There is not much we can do about it though unless aside from using another URL request library instead of `requests`, such as `urllib2`.

closed 2021-07-21 by dchassin