arras-energy/gridlabd-old #998: python solver receiving incorrect NR values

ebuech, 2021-09-13:
## Problem description

Learn and solve functions in python solver are receiving incorrect values from the Newton Raphson solver.

## Steps to reproduce

1) Launch an AWS EC2 instance (t2.xlarge) with the following GridLAB-D AMI:
hipas-gridlabd-4.2.21-210522-master/etc/beauharnois-21 (ami-0e81f427774d1042d)

2) Run the attached IEEE13 bus model (gridlabd IEEE13_new.glm from the command line) which will run for 10 timesteps
[ieee13_new_low.zip](https://github.com/slacgismo/gridlabd/files/7156467/ieee13_new_low.zip)

The code is set up to run newton raphson at each timestep followed by the learn function. The learn function prints out the constant power loads for each phase (SAr, SBr, SCr). 

You can observe that the SAr values are zero and SBr and SCr are identical. Based on the values in the player files, this is incorrect. Similar observations can be made from the voltage values.

Not included in the attached files: if you output a voltage prediction from the solve function and bypass newton raphson then it results in a segmentation fault and the simulation terminates.

## Expected behavior

Power values received by the solve and learn functions should match those in the player files.

## System information

* GridLAB-D version: HiPAS GridLAB-D 4.2.21-210522

dchassin, 2021-09-15:
@ebuech, can you add the output of the following command to the system information: `gridlabd --version=commit`?  Thx.

ebuech, 2021-09-15:
@dchassin Done

closed 2021-12-28 by dchassin