arras-energy/gridlabd-old #1049: Need to allow transformer turns ratios degradation

dchassin, 2022-01-03:
## Background

Many older transformers have degraded turns ratios.  

## Proposed solution

We need to allow the model to have a turns ratio degradation factor to account to burn-through as the transformer ages.

## Alternatives

None considered.

## Additional context

None.

dchassin, 2022-01-03:
@ftuffner, any thoughts about this idea?

ftuffner, 2022-01-04:
Seems like it would be relatively straight forward.  The big item will be in NR and making sure the admittance matrix gets updated on the "updated turns ratio/degradation", assuming you'll have this happen over time in a simulation.  It probably just will involve setting the `force_link_update` variable in module/powerflow/link.cpp or toggling a status change.

closed 2022-01-13 by dchassin