arras-energy/gridlabd-old #1066: AWS ec2 set environment during installation process

catzzz, 2022-01-07:
## Problem description
`PATH=/usr/local/bin:$PATH` cannot be set permanently by running ./install.sh. Without `PATH` variable during installation, users still can complete the installation correctly. However, `gridlabd` command cannot be performed directly under `gridlabd` folder after installation. Users have to find it in the build folder.

## Expected behavior
None
## System information
* GridLAB-D version:
* Operating system: AWS ec2

## Supporting information
none

dchassin, 2022-01-09:
This can be done by modifying the shells' `.rc` files.

dchassin, 2022-01-11:
@catzzz, can we close this or is this still something we need to fix?

catzzz, 2022-01-11:
Yes. this issue had been fixed.

catzzz, 2022-01-12:
Close issue

closed 2022-01-12 by catzzz